jerryscript-project/jerryscript #2884: Discussion: Thin C++ wrapper (jerry_value_t, core functions)

IngwiePhoenix, 2019-05-22:
I am currently constructing the layout for a C++ wrapper. This one is meant to be as thin and light as possible, as it's only purpose is to support OO programming with Jerry. Mainly meant to make wrapping C++ classes just a little bit easier.

https://gist.github.com/IngwiePhoenix/238db8d1a7f8731c6be7c4d28f20357c

The idea is:
- Use local variables instead of pointers to automatically trigger destruction (which should release the underlying `jerry_value_t`).
- Simplify defining native functions (`Value (*jerry_cxx_handler_cb)(Context)`) to make the handler signatures just a little less verbose. This is achieved by wrapping the actual callback in a generic `jerry_value_t (*)(jerry_value_t, jerry_value_t, jerry_value_t*, jerry_value_t)`  and assigning it as the native pointer in a function (setting the native pointer in an object).
- Allow quick conversion between the C++ types and C types (`operator jerry_value_t ();`) to allow using the regular C functions as expected.
- Mimic common methods - but only those that are _really_ needed.
- Use sub-classing and inheritance on a simple, non-diamond level, to allow other C++ types to derive from it (`class MyJerryVector: public JerryScript::Object, public MyVector`) so that going back and forth between native and scripted calls is easier.
- Try not to use templates. Templates are great, but can bring forth more confusion than clearance.

So far, I have tried to map out the basic types and their structure. I am going to create a few `inline` functions to allow JerryScript to have a better and more controlled way of dealing with the C++ wrappers (like returning a `JerryScript::Value` instead of a `jerry_value_t`). Also, I decided to create a very basic base class `Value`, which serves as the foundation for any other type.

What do you think of the structure and idea? Is there anything that I am missing?

My goal is to contribute this towards `jerry-ext` in the future, to allow a user to also use JerryScript in C++, if they like. The wrapper is meant to make a few things more OO-ish, but not to wrap the entire C API.

```c++
// For example:

use JerryScript::Object;
use JerryScript::Number;
use JerryScript::String;

Object o = Object::global();
o["foo"] = String("bar");
o["baz"] = Number(42);
```

Would be glad to hear your thoughts! :)

rerobika, 2019-10-18:
Hi @IngwiePhoenix,
First of all sorry for the late answer. The proposed idea looks promising, at the moment I do not see any blocking problem why it could not work. The only point which can create arguments is adding a C++ wrapper in `jerry-ext` properly, since JerryScript is fully written in C. Apart from that feel free to contribute/share us the the process, and hopefully we can find a suitable solution for it with the guidance of the reviewers.

rerobika, 2020-05-26:
Closing the issue due to inactivity, please reopen if needed.

closed 2020-05-26 by rerobika