jerryscript-project/jerryscript #2889: Question about native pointer.

acoinfo, 2019-05-29:
I used the object's properties and call object's method in native pointer `free_cb` callback, I found that there might be a memory leak or crash. test code as following, the native pointer `free_cb` will call `finalize()` method. 

Whether the native pointer does not allow use this object method or any property?

``` javascript
class Test {
	constructor() {
		this.a = 1;
	}
	
	finalize() {
		console.log('foo');
		this.a = 0;
	}
}

var t = new Test();
gc(true);

var t = new Test();
gc(true);

var t = new Test();
gc(true);

var t = new Test();
gc(true);
```

acoinfo, 2019-05-29:
I try another method: In the native pointer `free_cb` callback does not call the method of the this object, the program will not crash, but there is still a memory leak:

``` javascript
class Test {
	constructor() {
		this.a = 1;
		this.fd = fs.open('./xxx', fs.O_CREAT);
		
		Object.finalize(this, function() {
			console.log('adsfdsaf');
			if (this.fd >= 0) {
				fs.close(this.fd);
				this.fd = -1;
			}
		});
	}
}

var cnt = 0;

var t = new Test();
sys.gc(true);

var t = new Test();
sys.gc(true);

var t = new Test();
sys.gc(true);

var t = new Test();
sys.gc(true);

var t = new Test();
sys.gc(true);
```

rerobika, 2019-05-29:
@acoinfo 
> Whether the native pointer does not allow use this object method or any property?

The native pointers are only optional hidden properties, these do not affect the regular property access.

Could you please try to substitute the class language element with a normal function?

```js
function Test() {
    this.a = 1;
}
	
Test.prototype.finalize = function() {
    console.log('foo');
    this.a = 0;
}
```

The class language element is nearly a new feature, it would be great to know whether your issue exist without using it.

acoinfo, 2019-05-29:
There is still a memory leak, my native code is as follows:

``` C
/* destructor */
typedef struct jsre_object_destructor {
    jerry_value_t  jfunc;
    jerry_value_t  jthis;
} jsre_object_destructor_t;

/*
 * Object native info free.
 */
/*
 * Object native info free.
 */
static void jsre_object_destroy (jsre_object_destructor_t *dest)
{
    jerry_value_t jret;
    
    jret = jerry_call_function(dest->jfunc, dest->jthis, NULL, 0);
    jerry_release_value(jret);
    jerry_release_value(dest->jfunc);
    JSRE_FREE(dest);
}

/*
 * Object native information.
 */
static const jerry_object_native_info_t jsre_object_destructor_native_info = {
    .free_cb = (jerry_object_native_free_callback_t)jsre_object_destroy
};

/*
 * Object.finalize(obj, finalize)
 */
JS_FUNCTION(object_finalize)
{
    jerry_value_t jobj, jfinalize;
    jsre_object_destructor_t *dest;

    JS_CHECK_ARGS(2, object, function);
    jobj = JS_GET_ARG(0, object);
    jfinalize = JS_GET_ARG(1, function);

    dest = jsre_jval_get_object_native_handle(jobj, &jsre_object_destructor_native_info);
    if (dest) {
        jerry_release_value(dest->jfunc);

    } else {
        dest = (jsre_object_destructor_t *)JSRE_MALLOC(sizeof(jsre_object_destructor_t));
        jsre_jval_set_object_native_handle(jobj, dest, &jsre_object_destructor_native_info);
    }

    dest->jfunc = jerry_acquire_value(jfinalize);
    dest->jthis = jobj;

    return  (jerry_create_undefined());
}

/*
 * Object Destructor Extension achieve.
 */
void jsre_es_destructor (void)
{
    jerry_value_t jobject_cons = jsre_jval_get_property_global("Object");

    jsre_jval_set_method(jobject_cons, "finalize", object_finalize);

    jerry_release_value(jobject_cons);
}
```

So I need to register the finalize callback of a new object in the constructor, like this:
``` javascript
function Test() {
    this.a = 1;
    Object.finalize(this, function() {
        this.finalize();
    });
}

Test.prototype.finalize = function() {
    console.log('foo');
    this.a = 0;
}

var t = new Test();
```

After the program finishes running, jerryscript prompts for a memory leak.

galpeter, 2019-05-29:
When the `free_cb` is called the object on which the callback is registered is already destroyed (freed) thus you can not access it properties.

galpeter, 2019-05-29:
Could you check the result type of the `jerry_call_function` with `jerry_value_get_type`?

acoinfo, 2019-05-29:
Ok, I can use this method to avoid calling methods on objects.

``` javascript
function callback() {
    console.log('foo');
    this.a = 0;
}

function Test() {
    this.a = 1;
    Object.finalize(this, callback);
}

var t = new Test();
```
 jerryscript prompts for a memory leak.

If I use the arrow function:
``` javascript
function Test() {
    this.a = 1;
    Object.finalize(this, () => {
        console.log('foo');
        this.a = 0;
    });
}

var t = new Test();
```
Object cannot be recycled.

galpeter, 2019-05-29:
In the callback you are still accessing the `this` which is the free'd object.

I'm afraid the native pointer free callback was not really designed for calling jerry api methods, this `free_cb` is running during the gc phase. Potentially every js api calls could interfere with the mark-and-sweep gc throwing off the reference counting.

acoinfo, 2019-05-29:
Oh, if this is the principle, I suggest adding some notice in the document.

akosthekiss, 2019-06-28:
@galpeter @rerobika Has this been documented? Can we close this issue?

rerobika, 2019-06-28:
@akosthekiss I found a reference for this special case [here](https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/api/jerry.c#L111), but I think it should be mentioned in the public documentation as well.

PR #2932 (2019-07-02): Make API unavailable while processing native free callbacks
This patch fixes #2889 and extends the API documentation of the native free callbacks.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit a72d14f5 (2019-07-02): Make API unavailable while processing native free callbacks (#2932)

This patch fixes #2889 and extends the API documentation of the native free callbacks.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-07-02 by dbatyai