jerryscript-project/jerryscript #2891: Can Jerry delete a undefined object?

yeguixin, 2019-05-30:
When running the following code, I find Jerry cannot output error information.

```
var f = function(name){
    delete this.keywords[name];
    print("Hello Jerry");
}
f("jerry");
```
Outputs: Hello Jerry
I think jerry does not execute the code "delete this.keywords[name];". I don't know why this happens

rerobika, 2019-05-30:
@yeguixin IMO, the engine executes the delete operation as expected. The problem is that the `this.keywords` is `undefined` in this function scope and after that you trying to delete the `name` property of `undefined`. However accessing any property of `undefined` would raise `TypeError`, but the delete operation according to the standard checks whether the `object argument` is `undefined or not` and returns true if it is which is the correct behavior. 
Also you can test the result of the delete operation with a little modification:

```js
var f = function(name){
    print (delete this.keywords[name]); // true
    print("Hello Jerry");
}
f("jerry");
``` 
I hope this explanation was helpful and solves your problem :)

legendecas, 2019-05-30:
According to ecma-262 12.5.4.2, `delete UnaryExpression`
1. Let ref be the result of evaluating UnaryExpression.
2. ReturnIfAbrupt(ref).

Expression `this.keywords[name]` should abrupt in the case.

https://www.ecma-international.org/ecma-262/6.0/#sec-delete-operator-runtime-semantics-evaluation

rerobika, 2019-05-30:
@legendecas That's okay with an ES2015 guard, but how about ES5.1?

legendecas, 2019-05-30:
> @rerobika That's okay with an ES2015 guard, but how about ES5.1?

Looks like behavior is expected to be similar in both case since the UnaryExpression has to be evaluated.

https://www.ecma-international.org/ecma-262/5.1/#sec-11.4.1
11.4.1 The delete Operator
The production UnaryExpression : delete UnaryExpression is evaluated as follows:

1. Let ref be the result of evaluating UnaryExpression.
2. If Type(ref) is not Reference, return true.
...

rerobika, 2019-05-30:
@legendecas Indeed it looks similar, but there is a significant difference which is the missing `ReturnIfAbrupt` statement. Since the result of the evaluated `UnaryExpression` is an abrupt completion which IMO meets the requirement of `If Type(ref) is not Reference, return true.`. Therefore, the early return in the code for the undefined check only prevents the unnecessary exception raise/release. What do you think about it?

legendecas, 2019-05-31:
I'm confusing that es2015 was expected to be compatible with es 5.1. If es2015 throws an error on `delete undefined.foo` but es5.1 doesn't, there would be a compatibility issue.

I tested the case on chrome and safari, TypeError was thrown on both platforms.

yeguixin, 2019-06-21:
@legendecas Do you mean is this a compatibility issue of the ES standard? Generally, ES2016 is backward compatible.

akosthekiss, 2019-06-28:
The discussion on this issue has stopped but it seems it has not concluded.

@rerobika Should the ES5.1 and ES2015 profiles behave differently? If they should, do they? Does the reported test case behave in a non-standard way in any of the profiles (i.e., is this issue to be marked as an ES2015 bug)?

rerobika, 2019-06-28:
@akosthekiss According to my interpretation, the two standards are different at this point and the current implementation follows the requirements of ES5.1 standard. However, the new part of the ES2015 standard has not been implemented yet, but IMO we can treat this issue as a feature request.

PR #3312 (2019-11-14): Improve delete property with undefined base
This patch finally resolves #2891.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit be95aa33 (2019-11-14): Improve delete property with undefined base (#3312)

This patch finally resolves #2891 also the removes the related bytecode since it has become unnecessary.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-11-14 by dbatyai