jerryscript-project/jerryscript #2900: Regular expression problem.

cyongbin, 2019-06-10:
- Jerryscript's regular expression not support positive/negative assert, em:
```
/(?<=\?)[^#]+/
```

LaszloLango, 2019-06-11:
JerryScript supports the positive and negative assertions in RegExp objects. Your testcase is not a valid assertion pattern.

cyongbin, 2019-06-11:
> JerryScript supports the positive and negative assertions in RegExp objects. Your testcase is not a valid assertion pattern.

- My testcase is ok, i had test it in node.js env.
- node.js em:
```
var url =  @'http://test.com?name=tom';
var quary = url.match(/(?<=\?)[^#]+/);
   @@@console.log(quary);
```
- It can only write as below in jerry:
```
var url = 'http://test.com?name=tom';
var quary = url.match(/\?[^#]+/);
query = query ? query[0].splice(1) : null;
console.log(quary);
```

zherczeg, 2019-06-11:
We have lookahead/lookbehind and positive/negative assertions in regexps. ES5 only supports lookahead and your pattern has lookbehind.

cyongbin, 2019-06-12:
Thank you. I look forward to your realization of lookbehind model.

akosthekiss, 2019-06-28:
> We have lookahead/lookbehind and positive/negative assertions in regexps. ES5 only supports lookahead and your pattern has lookbehind.

@zherczeg Does that mean that the script with this regex pattern should work if the engine is built with ES2015 profile? (<-- "We have ... lookbehind" + "ES5 only supports lookahead" + "your pattern has lookbehind")

zherczeg, 2019-07-02:
Lookbehind is introduced in ES2018:
https://www.ecma-international.org/ecma-262/9.0/#prod-Assertion

Was not in ES2017:
https://www.ecma-international.org/ecma-262/8.0/#prod-Assertion

So only ES2018 compatible engines should support it.

akosthekiss, 2019-07-02:
I've edited the issue title to be more to the point. I'm also marking the issue as a feature request but closing it at the same time as ES8/9 is out of scope for the project for now. We should get full ES6 support landed first.

closed 2019-07-02 by akosthekiss