jerryscript-project/jerryscript #2905: The try-catch block is unable to catch the exception?

YaoHouyou, 2019-06-12:
The try-catch block is unable to catch the error when I execute the code shows below, but throw  `ERR_REF_COUNT_LIMIT` error .
````
var func = function () {
	foo.prototype = new Array(1, 2, 3);
	function foo() {}
	var f = new foo();
	f.length = f;
	try {
		print("run to try");
		print("Using f will give an error: " + f);
		print("run to 2");
	} catch (e) {
		print("run to catch!")
	}
};
func();
````
The output is:
````
run to try
Error: ERR_REF_COUNT_LIMIT
````

LaszloLango, 2019-06-12:
`ERR_REF_COUNT_LIMIT` is a fatal error of the engine and not an ECMA262 standard exception, so you cannot catch it.

YaoHouyou, 2019-06-12:
Do you mean it's a bug of jerry?

LaszloLango, 2019-06-12:
No, it is not a bug.

rerobika, 2019-06-13:
@YaoHouyou I agree with @LaszloLango It's not a bug, it is simply a JS call stack overflow. If you remove the `try-catch` block around the `print("Using f will give an error: " + f);` and test it in any browser you will get a RangeError (This is what the catch clause truly catches). Jerry does not manages the call stack size internally, which could lead to `ERR_REF_COUNT_LIMIT` since each call adds an extra reference to the function object/arguments, and in a while, these continuously growing references reaches the limit which terminates the engine.

akosthekiss, 2019-06-28:
@rerobika jerry-core can be configured to have a recursion limit via the `VM_RECURSION_LIMIT` cmake option (or via the `--vm-recursion-limit` option of build.py). Would that solve the issue and make the engine throw/catch RangeError correctly?

rerobika, 2019-06-28:
@akosthekiss IMO after a little modification of  `VM_RECURSION_LIMIT` it would be suitable for this problem. This guard prevents only the recursion of the interpreted code blocks but the builtin/native functions calls are not guarded either. I'll open a PR for it.

PR #2935 (2019-07-05): Limit the call stack size for native/builtin functions as well
VM_RECURSION_LIMIT only prevented the recursion of interpreted codeblocks but
non-interpeted codeblocks can also create stack overflow due to the too deep recursion.

This patch fixes #2905.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit e902b870 (2019-07-05): Limit the call stack size for native/builtin functions as well (#2935)

VM_RECURSION_LIMIT only prevented the recursion of interpreted codeblocks but
native/builtin function calls can also create stack overflow due to the too deep recursion.

This patch fixes #2905.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-07-05 by rerobika