jerryscript-project/jerryscript #2906: Can Jerry report the Reference error when parsing JS code?

yeguixin, 2019-06-12:
###### Jerry version:

> Checked version:  351acdf

Build command:  python tools/build.py --debug --error-messages=on --profile=es2015-subset

 ###### Test case:

```
  1 var foo = function () {
  2         this.timer_ = null;
  3         if (this.shouldFire_ && !this.pauseCount_) {
  4                 this.timer.onstart();
  5                 800 = false;
  6
  7         }
  8 };
  9
 10 foo();
```

When run the above testcase, Jerry cannot report the Reference error (line 5) while other JS engines such as V8, SpiderMonkey can.

dbatyai, 2019-06-12:
Static semantics were introduced in ES2015, which is why engines implementing the newer standard will have additional checks while parsing.

JerryScript is primarily an ES5.1 engine, where it is only required to throw a ReferenceError when the assignment actually happens.

Since the assignment in question is in an unreachable code path, it will never result in a ReferenceError. However, if you modify the test case to make the code reachable, then an error will be thrown when the assignment is executed, as expected by the ES5.1 standard.

akosthekiss, 2019-06-28:
The discussion on this issue seems to have stopped but I'm not sure it has concluded.

@dbatyai @rerobika Is static semantics mandated by ES2015? The issue report seems to include the build command with `es2015-subset` set as the profile. If static semantics is mandated by the standard, can it be implemented in the engine/parser as we move towards a more complete ES2015 support? Shall we turn this issue report into a feature request?

rerobika, 2019-06-28:
@akosthekiss I think we can treat this issue as a feature request, since this static semantics is already provided by the parser just the error is emitted to the bytecode stream instead of thrown.

akosthekiss, 2019-06-28:
> static semantics is already provided by the parser just the error is emitted to the bytecode stream instead of thrown

@rerobika But which approach is correct? The answer has to come from the specification. We shouldn't implement it just because other engines work this way. But if ES2015 mandates it, then it's a valid feature request.

rerobika, 2019-07-01:
@akosthekiss For the ES5.1 profile the error should be emitted to the bytecode stream which creates `ReferenceError` during the execution and the engine does exactly the same.

For the ES2015 profile the error must be thrown as a SyntaxError during the parsing but the Jerry still emits the `CBC_THROW_REFERENCE_ERROR` opcode to the bytecode stream. This should be implemented in the future.

rerobika, 2020-05-26:
This issue is going to be resolved by #3742 .

rerobika, 2020-07-06:
Resolved by #3742.

closed 2020-07-06 by rerobika