jerryscript-project/jerryscript #2914: An error should be thrown when defining a property for a read-only variable in strict mode.

Implementist, 2019-06-17:
As the title suggests, the testcase is:
```
var func = function (number) {
    "use strict";
    number.snippetText = "";
};
func(-334918463);
```
The defination and assignment at the third line leads to an exception when I run this testcase in V8 and SpiderMonkey. But Jerry doesn't throws any exception.

rerobika, 2019-06-18:
@Implementist Thanks for the report I think it is a valid bug, since I've found the corresponding part of the standard for that special case.

https://www.ecma-international.org/ecma-262/5.1/#sec-8.7
```
The object that may be created in step 1 is not accessible outside of the above method.
An implementation might choose to avoid the actual creation of that transient object.
The only situations where such an actual property assignment that uses this internal
method can have visible effect are when it either invokes an accessor function
or is in violation of a Throw predicated error check. When Throw is true any property
assignment that would create a new property on the transient object throws an error.
```

Do you have any fix for this issue or are you planning to open a PR for this problem? (Ofc, it is only optional.)

Implementist, 2019-06-18:
Thanks for your reply. I have no idea about how to fix this at present.

rerobika, 2019-06-18:
@Implementist No problem I'll take a look at it.

galpeter, 2019-06-18:
@rerobika I've done a bit of debugging and created this: https://github.com/galpeter/jerryscript/commit/d3f6254e856a0053baf84f96c63bee919179e38c  however I think it is only a partial fix.

PR #2917 (2019-06-24): Ensure that the constructed object in [[Set]] operation with primitive base cannot be extensible
Related part of the standard: https://www.ecma-international.org/ecma-262/5.1/#sec-8.7 1st note.

This patch fixes #2914.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 44b21de7 (2019-06-24): Ensure that the constructed object in [[Set]] operation with primitive base cannot be extensible (#2917)

Related part of the standard: https://www.ecma-international.org/ecma-262/5.1/#sec-8.7 1st note.

This patch fixes #2914.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-06-24 by rerobika