jerryscript-project/jerryscript #2924: jerryscript.net - invalid SSL certificate

trivikr, 2019-06-24:
### Steps to reproduce
https://jerryscript.net/ shows invalid SSL certificate
This issue was noticed while adding link to JerryScript in OpenJSF CPC README in https://github.com/openjs-foundation/cross-project-council/pull/218 

### Expected output
The website https://jerryscript.net/ loads without any certificate issues

### Actual output
https://jerryscript.net/ shows invalid SSL certificate

trivikr, 2019-06-26:
Any update on this request?

akosthekiss, 2019-06-27:
@trivikr Thanks for reporting this issue. It seems that it is related to outdated IP addresses (https://help.github.com/en/articles/troubleshooting-custom-domains#https-errors). We have let our DNS maintainers know about it, let's hope for a fix soon.

akosthekiss, 2019-07-04:
It's expected to get solved in some days.

akosthekiss, 2019-07-09:
The issue has been solved. Our GH pages work OK over https.

closed 2019-07-09 by akosthekiss