jerryscript-project/jerryscript #2930: Strange performance issue

weixiongmei, 2019-06-27:
[image]
Wrap the code in a function, the code above took about 2 seconds.

[image]
But if just executing the code in the second picture above, it took 13 seconds to finish, its 6.5x slower than wrapping it into the function....

Any thought? thanks~~~

akosthekiss, 2019-06-27:
@weixiongmei Please, post code snippets using markdown fenced code block syntax. Avoid posting code as images (screenshots). Moreover, the issue you are reporting definitely needs context: `console.log` is not a JerryScript entity. Provide a test case that can reproduce the issue using JerryScript only (e.g., using its command line tool `jerry`). Also, please, disclose how you are using JerryScript, how you are compiling it, what platform is it running on, etc.

weixiongmei, 2019-06-27:
Hi @akosthekiss , the console.log is the custom module, which it's running the latest jerryscript on ESP32,  please just test the code and print the result output in your own way to see the time consumtion.. Thanks~~

akosthekiss, 2019-06-27:
Reproduction follows:

- Platform: Ubuntu 18.04.2 LTS (GNU/Linux 4.15.0 x86_64)
- Git revision: f6f52a2b
- Build command: `tools/build.py --clean`
- Test case (`x.js`):
```js
//(function() {
    print("Start Benchmark");
    while(true) {
        var st = new Date();
        var k = 0;
        for(var i = 0; i < 1000; i++) {
            for(var j = 0; j < 1000; j++) {
                //k += j;
            }
        }
        var et = new Date();
        var ct = (et- st);
        print("Time:" + ct.toString() + " Count:" + k.toString());
    }
//})();
```
- Execute as: `build/bin/jerry x.js`
- Output with first and last lines commented out: `Time:58 Count:0`
- Output with active first and last lines: `Time:23 Count:0`
- Slowdown factor is about 2.5x.

weixiongmei, 2019-06-27:
Even though you only got 2.5x slower, but still a huge difference. wondering why it's much faster by just wrapping the code into the function. Hope it will be fixed soon. Thank you so much~~

rerobika, 2019-06-28:
@weixiongmei The reason is that when your code is wrapped to a function jerry can use register optimization but if you place it into the global scope, this kind of optimization cannot be performed. Most of the functions do not export their variable bindings therefore they can be stored in a register file and can be accessed via a register index which is significantly faster than storing them in lexical environment. Unfortunately the global scope's bindings must be stored in lexical environment since any function can perform the `eval` operation (which also prevents the register optimization of a function) and  during the variable binding lookup the global lexical environment bindings **must** be available and this  can only be ensured by storing the bindings in lexical environment.

I hope this explanation was helpful.

akosthekiss, 2019-06-28:
It seems to me that this is not a performance regression in the global scope case but a performance improvement in the function scope case, thanks to an optimisation possibility that opens up there. It also seems that such an optimisation is not possible at the global scope, which means that this performance difference will persist between the two approaches.

closed 2019-06-28 by akosthekiss