jerryscript-project/jerryscript #2936: UTF-8 variable name error

weixiongmei, 2019-07-01:
When using UTF-8 variable name, the "jerry_parse" function will returns a value "1073429487", wondering how to get it fixed?

```javascript
var 测试 = "您好";
```

akosthekiss, 2019-07-01:
It's highly preferred to submit version info and platform info, as well as build and execution commands along issue reports to make them reproducible. Moreover, 1073429487 probably doesn't mean a thing,  as `jerry_parse` returns a `jerry_value_t`, which should _not_ be interpreted as a number.

Anyway, reproduction follows with necessary information provided:
```sh
$ uname -mrs
Darwin 18.6.0 x86_64

$ tools/build.py --clean --debug --error-messages=on --logging=on

$ build/bin/jerry --version
Version: 1.0 (dea73d87)

$ build/bin/jerry ../unicodeid.js 
var 测试 = "您好";
~~~~^
Script Error: SyntaxError: Identifier expected. [line: 1, column: 5]
```

rerobika, 2019-07-02:
@weixiongmei Thanks for the report, the issue seems valid on the current master. We are going fix this issue as soon as possible.

weixiongmei, 2019-07-02:
@rerobika Thank you so much, it would be very helpful for the people who may use UTF-8 to name the variables or functions in the js code. Also, i apologize for not reporting in the specified format, because the port of ESP32 that i'm using can't print the error message....

weixiongmei, 2019-07-03:
@akosthekiss Hi, Can you tell how did you print the error message? I enabled the JERRY_ERROR_MESSAGES, I can manually add "prinf" to the "js-parser.c" for the error message. But, i belive it's not the right way to do it, wondering how did you print the error message from the return_value of "jerry_parse or jerry_run".

```c++
#if ENABLED (JERRY_ERROR_MESSAGES)
    const lit_utf8_byte_t *err_bytes_p = (const lit_utf8_byte_t *) parser_error_to_string (parser_error.error);
    lit_utf8_size_t err_bytes_size = lit_zt_utf8_string_size (err_bytes_p);

    ecma_string_t *err_str_p = ecma_new_ecma_string_from_utf8 (err_bytes_p, err_bytes_size);
    ecma_value_t err_str_val = ecma_make_string_value (err_str_p);
    ecma_value_t line_str_val = ecma_make_uint32_value (parser_error.line);
    ecma_value_t col_str_val = ecma_make_uint32_value (parser_error.column);

    printf("%s [line: %d, column: %d]",err_bytes_p, parser_error.line, parser_error.column);

    ecma_value_t error_value = ecma_raise_standard_error_with_format (ECMA_ERROR_SYNTAX,
                                                                      "% [line: %, column: %]",
                                                                      err_str_val,
                                                                      line_str_val,
                                                                      col_str_val);

    ecma_free_value (col_str_val);
    ecma_free_value (line_str_val);
    ecma_free_value (err_str_val);

    return error_value;
#else /* !ENABLED (JERRY_ERROR_MESSAGES) */
```

akosthekiss, 2019-07-03:
@weixiongmei There should be no need to modify js-parser.c . I've used the command line utility to execute the test, which contains logic to print info about unhandled exceptions that happen during parsing or running a script. Everything happens using public API functions. Please, have a look at jerry-main/main-unix.c, especially at the `print_unhandled_exception` function and its call sites. That function is not part of either jerry-core or jerry-ext as it contains some code specific to jerry-main and as such it is not fully reusable (at least, at for now), but you may copy it and tailor it to your needs.

weixiongmei, 2019-07-03:
@akosthekiss Hi, i just using the code you posted in the github a long time ago for printing the error message, but it seems like doesn't work.
```c++
void print_string_form (jerry_value_t val)
{
  jerry_value_t str_val = jerry_value_to_string (val);
  jerry_size_t str_size = jerry_get_string_size (str_val);

  jerry_char_t str_buf[str_size + 1];
  jerry_string_to_char_buffer (str_val, str_buf, str_size);

  jerry_release_value (str_val);

  printf ("Error: %s\n", str_buf);
}

void app_main()
{
  char *entry = "/spiffs/index.js";
  size_t size = 0;
  jerry_char_t *script = jerry_port_read_source(entry, &size);
  jerry_value_t parse_code = jerry_parse((jerry_char_t *)entry, strlen(entry), script, size, JERRY_PARSE_NO_OPTS);
if (jerry_value_is_error(parse_code)) {
  	print_string_form(parse_code);
}
jerry_value_t ret_value = jerry_run(parse_code);
print_string_form(ret_value);
jerry_release_value(ret_value);
jerry_release_value(parse_code);
jerry_port_release_source(script);
}
```

weixiongmei, 2019-07-03:
@akosthekiss "print_unhandled_exception" is not working neither, the err_str_size returns 0.

akosthekiss, 2019-07-03:
@weixiongmei The following minimal example works for me:
```c
#include <stdio.h>

#include "jerryscript.h"

static void
print_exception (jerry_value_t error_value) /**< error value */
{

  jerry_value_t err_str_val = jerry_value_to_string (error_value);
  jerry_size_t err_str_size = jerry_get_string_size (err_str_val);
  jerry_char_t err_str_buf[err_str_size + 1];

  jerry_size_t string_end = jerry_string_to_char_buffer (err_str_val, err_str_buf, err_str_size);
  err_str_buf[string_end] = 0;

  printf ("Error: %s\n", err_str_buf);
  jerry_release_value (err_str_val);
} /* print_exception */

int main (void)
{
  jerry_init (JERRY_INIT_EMPTY);

  const jerry_char_t bugous_src[] = "{";

  jerry_value_t ret_value = jerry_create_undefined ();
  ret_value = jerry_parse (NULL, 0,
                           bugous_src, sizeof (bugous_src) - 1,
                           JERRY_PARSE_NO_OPTS);

  if (jerry_value_is_error (ret_value))
  {
    ret_value = jerry_get_value_from_error (ret_value, true);
    print_exception (ret_value);
  }

  jerry_release_value (ret_value);

  return 1;
} /* main */
```
Its output is `Error: SyntaxError: Primary expression expected. [line: 1, column: 2]`.

The `ret_value = jerry_get_value_from_error (ret_value, true);` line is an important difference compared to old code. The API around errors have changed somewhat (around #2350). See http://jerryscript.net/api-reference/#jerry_get_value_from_error . That may also affect your experience with `print_unhandled_exception`. 

Do take a look at main-unix.c. It is always a good starting example.

weixiongmei, 2019-07-03:
@akosthekiss Thank you so much, your minial example works like a champ now.. Hopefully the UTF-8 variable name issue would be fixed very soon too..

weixiongmei, 2019-07-05:
@rerobika Can you please also fix the UTF-8 for the magic string too? The magic string does not supports UTF-8... Thanks ~~~

rerobika, 2019-07-05:
@weixiongmei The magic strings only accept ASCII characters, but you can use the external magic strings if you need UTF-8 characters. Please check the (jerry_register_magic_strings)[https://github.com/jerryscript-project/jerryscript/blob/8766bb3b379a26ca21014ca7412c0100fc1853ae/jerry-core/api/jerry.c#L275] API function.

weixiongmei, 2019-07-05:
@rerobika Is it possible to add UTF-8 support for magic string? with the jerry_register_magic_strings, can i bind it to a function? I need this because i want to add some native methods/properties to all the BUILDIN types with UTF-8 name. I want to achieve something like "BuildInTypes.toString()" but in UTF-8. Thanks

rerobika, 2019-07-05:
@weixongmei Sorry, but I do not understand exactly what you would like to achieve. Would you like to add a new routine with an UTF-8 name for e.g. String.prototype at compile time (like toStri g, replace, split etc.) or after the initialization of the engine you would like to register native functions for built-in objects with an UTF-8 property name with `jerry_set_property`?

weixiongmei, 2019-07-05:
@rerobika Yes, i would like to add new routine with an UTF-8 name. For instance, all the Built-In objects can call *.toString() but i want to add another method like *.toString(), but in UTF-8 like *.UTF-8(). How can i do it? Thanks~~

rerobika, 2019-07-05:
@weixiongmei Thanks for the explanation. Routine names MUST be magic strings and those are must be an ASCII character sequence. The project highly depends on these magic strings so changing it is not recommended.

But, I think there is a better solution for this problem (the code is only a snippet without being precise)Ű

```c
jerry_value_t
my_routine_handler (const jerry_value_t func_obj_val, /**< function object */
                    const jerry_value_t this_p, /**< this arg */
                    const jerry_value_t args_p[], /**< function arguments */
                    const jerry_length_t args_cnt) /**< number of function arguments */
{
  // Do the expected routine's behavior here
  return jerry_create_undefined();
}

jerry_value_t
handler_register (jerry_value_t object, /**< object to register the routine */
                  const jerry_char_t *name_p, /**< name of the function */
                  jerry_external_handler_t handler_p) /**< function callback */
{
  jerry_value_t function_name_val = jerry_create_string_from_utf8 (name_p);
  jerry_value_t function_val = jerry_create_external_function (handler_p);

  jerry_value_t result_val = jerry_set_property (object, function_name_val, function_val);

  jerry_release_value (function_val);
  jerry_release_value (function_name_val);

  return result_val;
}

void register_utf8_handlers() {
  jerry_value_t global_obj_val = jerry_get_global_object ();

  jerry_value_t string_builtin = jerry_get_property (global, "String");
  jerry_value_t string_prototype_builtin = jerry_get_property (string_builtin, "prototype");

  handler_register (string_prototype_builtin, "<any UTF-8 sequence here>", my_routine_handler)

  jerry_release_value (string_builtin);
  jerry_release_value (string_prototype_builtin);
  jerry_release_value (global_obj_val);
}

int main () {
  jerry_init(JERRY_INIT_EMPTY);

  register_utf8_handlers();
  // Now, you can access String.prototype['<any UTF-8 sequence here>()'] which will call the my_routine_handler function
  // run the engine, 

  jerry_cleanup();
  return 0;
}

```

I hope this will help for you.

akosthekiss, 2019-07-05:
@weixiongmei To clear things up a bit:
- "Magic strings" are completely internal implementation details of the engine. They are ASCII and will remain ASCII as that's both sufficient and memory efficient to implement the specification (all spec-mandated built-in object and function names are ASCII only). If you want to change that, you are on your own. You may open a PR if you have an implementation, but it will have to be as memory and performance efficient as the current code base to get accepted. I'd not suggest to take this path.
- "External magic strings" are available from the API and they already support UTF-8. So, if you embed the engine in your own project and you'd like to have objects or functions with non-ASCII names, plus you'd like to make use of the memory efficiency provided by the external magic strings, you can already do so.
- And it also works without external magic strings, as shown in the above example. You don't need to register external magic strings with the engine to have non-ASCII-named objects and functions available. (Although that will consume more memory.)

To sum up: if you use the public API, you can add objects/functions/properties with arbitrary UTF-8 names, and those UTF-8 strings may be registered with the engine as external magic strings to save memory.

closed 2019-07-05 by dbatyai

weixiongmei, 2019-07-05:
@rerobika @akosthekiss If I really want to add the UTF-8 support to the magic string, which files should i look at, will it be extremely difficult? I'm trying to midify the **lilt-magic-strings.ini file** and regenerate the **lilt-magic-strings.ini.h**, Anything with a name LIT_MAGIC_STRING_* works with UTF-8, but anything that has U or UL at the end of the name (Such as LIT_MAGIC_STRING_*_UL) will not works. I need to add the UTF-8 supports for the magic string is because i'm trying to replace all the strings in the **lilt-magic-strings.ini file** to another language which required UTF-8 support. The performance & memory is not an issue, because i'm developing something for the child educations, and those little kids don't understand in English. I can pay in $USD for this feature. Would you two like to help? If so, can you contact me via e-mail? Thank you~~~

weixiongmei, 2019-07-06:
@rerobika @akosthekiss Is there anyway to re-name the Built-In Object/Class/Method after the engine is compiled? For instance, change the name of "Date" to "测试" after or before **jerry_init** is called, can **jerry_set_property** or any API can do that? Thanks~~

weixiongmei, 2019-07-07:
@rerobika @akosthekiss I fixed it

PR #2944 (2019-07-05): Improve gen-unicode.py to support unicode ranges
This patch fixes #2936

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 2b8c4286 (2019-07-05): Improve gen-unicode.py to support unicode ranges (#2944)

This patch fixes #2936

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu