jerryscript-project/jerryscript #2945: Are these Syntax supported in the current master branch?

weixiongmei, 2019-07-05:
Hi, wondering are theses Syntax "private, public, let, get, set" supported in the current master branch? Or where can i find the information for which Syntax is supported? Would like to create my own class with some private methods, don't know how to do with Jerryscript.
Please HELP, thank you!!

rerobika, 2019-07-05:
@weixiongmei Please check this [list](https://github.com/jerryscript-project/jerryscript/blob/8766bb3b379a26ca21014ca7412c0100fc1853ae/jerry-core/api/jerry.c#L275) of the currently supported ES2015 features. Unfortunately, most of your requested features are not available at the moment.

akosthekiss, 2019-07-05:
Two comments here:
- @rerobika the link to the list of supported ES2015 features is off, I think.
- @weixiongmei all keywords are recognized by the JS lexer but not all of them are supported yet. But: `finally` has been there for a long time (it's ES5.1, right?); `get` and `set` are supported in the ES2015 (subset) profile implementation; `private`, `public`, `protected`, and `await` are future reserved words in the ES2015 spec so the implementation is OK not supporting them; which leaves us with `let` and `const` from the above list as the only really unsupported features (as far as the ES2015 spec is concerned).

akosthekiss, 2019-07-05:
A bit more on the private class members topic: JavaScript has no such concept, not even the latest spec version. There is a _proposal_ for private fields but that's only a proposal for now: https://github.com/tc39/proposal-class-fields , and it is not even based on the `private` keyword but has a hashtag-based syntax.

JerryScript does not take on the task of implementing proposals that are not in the supported spec versions.

rerobika, 2019-07-08:
@weixiongmei Sorry for the inconvenience, I've updated the link with the currently supported ES2015 features.

weixiongmei, 2019-07-09:
@rerobika ,Thank you so much, one more question, 
```C++
jerry_init (JERRY_INIT_SHOW_OPCODES | JERRY_INIT_SHOW_REGEXP_OPCODES | JERRY_INIT_MEM_STATS | JERRY_INIT_MEM_STATS_SEPARATE);
```
Why there's no output for opcodes & memory status. Do i have to call any function to get those info? Also, i'm trying to use snapshot, but couln't get it works... Can you give a minial example code for that if possible? thanks a lot~~~

akosthekiss, 2019-07-09:
@weixiongmei This last question seems to be unrelated to the original topic of ES2015 classes. Could you please open a new issue for the new topic?

Furthermore, if you are OK with the answers you got for the original question posted here, please signal it with closing this issue.

closed 2019-07-09 by weixiongmei