jerryscript-project/jerryscript #2946: BUG: Date.parse()

cyongbin, 2019-07-05:
var timestamp = Date.parse('Wed, 03 Jul 2019 03:15:44 GMT');
console.log(timestamp);
// Result: NaN
// Expected: 1562123744000

LaszloLango, 2019-07-05:
Indeed, the current version does not handle all of the date format strings. According to the standard the following methods must work:

```javascript
Date.parse(new Date().valueOf());
Date.parse(new Date().toString());
Date.parse(new Date().toISOString());
Date.parse(new Date().toUTCString());
```

Output:
```
jerry> Date.parse(new Date().valueOf());
NaN
jerry> Date.parse(new Date().toUTCString())
NaN
jerry> Date.parse(new Date().toString())
NaN
jerry> Date.parse(new Date().toISOString())
1562320786073
```

akosthekiss, 2019-07-05:
A bit more precisely: the following expressions should produce the same numeric value:
```js
// var x = new Date()
x.valueOf()
Date.parse(x.toString())
Date.parse(x.toUTCString())
Date.parse(x.toISOString())
```
The `valueOf` line works correctly:
```
jerry> var d = new Date()
jerry> d.valueOf()
1562328810063
jerry> Date.parse(d.toUTCString())
NaN
jerry> Date.parse(d.toString())
NaN
jerry> Date.parse(d.toISOString())
1562328810063
```
However, the spec mandates (ES5.1 15.9.4.2) that the implementation of `Date.parse` has to deal with string values that could be produced by `toString` or `toUTCString` as well. That's where we fail:
```
jerry> d.toUTCString()
Fri, 05 Jul 2019 12:13:30 GMT
jerry> d.toString()
Fri Jul 05 2019 14:13:30 GMT+02:00
jerry> d.toISOString()
2019-07-05T12:13:30.063Z
```
There are two ways to fix the problem in a spec-conforming manner: either `Date.parse` has to be extended to parse the two additional formats or `toString` and `toUTCString` have to be made identical (or almost identical) to `toISOString` (see 15.9.5.2 and 15.9.5.42).

Note: if we choose the second option, the originally reported script will still result NaN, but correctly so!

cyongbin, 2019-07-12:
I will try the second way. Thank you.

ossy-szeged, 2019-08-02:
I prefer the first way, because other JS engines use same format for toString() and toUTCString(). (tested with JSC, Firefox, Chrome, Edge) So we shouldn't change any output format, but make the parser to be able to parse toString() and toUTCString() format too.

Recently I found a related issue (wrong output if the year < 0 or year >9999). I'm going to fix that issue soon and will check how can we improve Date.parser accordingly.

PR #3314 (2019-11-26): Make Date.parse() ECMA-262 conform
Changes:
- Parse output of Date.prototype.toString() and Date.prototype.toUTCString()
- Date.prototype.toString() is ECMA-262 v9 conform now, only TZ part changed
  (Before ECMA-262 v9 it was implementation-dependent.)
- Reused day_names_p and month_names_p arrays (and made them more efficient)
- Tests updated and new tests added

Fixes #2946.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác oszi@inf.u-szeged.hu
commit 4d422e17 (2019-11-26): Make Date.parse() ECMA-262 conform (#3314)

Changes:
- Parse output of Date.prototype.toString() and Date.prototype.toUTCString()
- Date.prototype.toString() is ECMA-262 v9 conform now, only TZ part changed
  (Before ECMA-262 v9 it was implementation-dependent.)
- Reused day_names_p and month_names_p arrays (and made them more efficient)
- Tests updated and new tests added

Fixes #2946.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác oszi@inf.u-szeged.hu
closed 2019-11-26 by dbatyai