jerryscript-project/jerryscript #2949: problem about regular expression

YaoHouyou, 2019-07-08:
### Versions: `b6fc4e1` `e944cda`

### Testcase:
````
1. var foo = function () {
2.         var reg1 = new RegExp("{");
3.         print(reg1);
4.         var reg2 = new RegExp("A{");
5.         print(reg2);
6. }
7. foo();
````
### Command:
`./jerryscript/build/bin/jerry testcase.js`

### Result:
````
/{/
Exception backtrace:
  0: testcase.js:4
  1: testcase.js:7
Script Error: SyntaxError: invalid quantifier
````
### Description:
Why does jerry report an error on line 4 but not on line 2? Maybe, jerryscript should not throw an exception on line 4.

galpeter, 2019-08-01:
If I read the on the ecma standard correctly the `/{/` regexp is not valid ( see the grammar: https://www.ecma-international.org/ecma-262/6.0/#sec-patterns).

However! There is an annex B which describes "Additional ECMAScript Features for Web Browsers". This section contains extensions for the regexp grammar also ( https://www.ecma-international.org/ecma-262/6.0/#sec-regular-expressions-patterns ) and this grammar allows parsing the `/{/` regexp.

In JerryScript there is a `JERRY_REGEXP_STRICT_MODE` option for better compatibility with other engines and allow parsing a bit more than the grammar described in the first case. As this option is disabled by default the `{` pattern  will be parsed. Still the full annex B support is not there which would allow parsing the `A{` pattern.

If the `JERRY_REGEXP_STRICT_MODE` is enabled (`./tools/build --regexp-strict-mode off`) both of the patterns would be invalid and an error would be thrown.

rerobika, 2019-10-18:
@YaoHouyou pinging the issue, since there was no activity in the last 2,5 months. If I'm not mistaken after @galpeter 's response the problems seems to be resolved. Are you ok the close this issue?

closed 2019-10-18 by YaoHouyou