jerryscript-project/jerryscript #2957: Regular object can only capture 567 bytes buffer.

cyongbin, 2019-07-10:
- Regular expression objects can only capture 567 bytes of content. Is this a bug or design  so?
- em.
```js
var str = `
<!--560+7-->
<!--
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
0123456789
-->

<!--561+7-->
<!--
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890
-->
`;

var comment = /<!--(?:[^-]|(?:-(?!->)))*-->/g;
str = str.replace(comment, function(tag, index){
	console.log(index, '-', tag);
	return `<tag index=${index}/>`;
});
console.log(str);
```

- result:
```
0 - <!--560+7-->
14 - <!--
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
01234567890123456789012345678901234567890123456789
0123456789
-->
611 - <!--561+7-->
error: FATAL ERROR: thread 401001b[regular.js] stack overflow.
```

cyongbin, 2019-07-10:
I tried with another preferred regular expression and I have not encountered the above problem for a while, but I am not sure.
```
var comment = /<!--(?:[^-]*|(?:-(?!->)))+-->/g;
```

rerobika, 2019-11-13:
Hi @cyongbin ,

There were a lot of improvement about the regexp engine (#3136) and the string replace (#3284). Currently I've tested the reported issue and the stack overflow problem has disappeared. IMO the we can treat the issue as resolved, please feel free to reopen it anytime.

closed 2019-11-13 by rerobika