jerryscript-project/jerryscript #2963: JSON.strigify with replacer = Array causes segmentation fault

dodgerblue, 2019-07-11:
Jerry version:
```
Checked revision: c304b9a3

Build command: ./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
                --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
                --compile-flag=-fno-common --compile-flag=-g --strip=off \
                --system-allocator=on --logging=on --linker-flag=-fuse-ld=gold \
                --error-messages=on --profile=es2015-subset
```
OS:
```
Linux 4.18.0-25-generic #26~18.04.1-Ubuntu SMP \
Thu Jun 27 07:28:31 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```
Test case - test_03.js:
```
JSON.stringify(10, Array)
```
Backtrace:
```
$ ./build/bin/jerry /share/js_validation/test_03.js
ASAN:DEADLYSIGNAL
=================================================================
==98861==ERROR: AddressSanitizer: stack-overflow on address 0xff364f20 (pc 0xf79c9ea8 bp 0xff365348 sp 0xff364f00 T0)
    #0 0xf79c9ea7 in malloc (/usr/lib32/libasan.so.4+0xe5ea7)
    #1 0x566192c1 in jmem_heap_alloc_block_internal /share/repos/jerryscript/jerry-core/jmem/jmem-heap.c:293
    #2 0x56619391 in jmem_heap_gc_and_alloc_block /share/repos/jerryscript/jerry-core/jmem/jmem-heap.c:327
    #3 0x56619466 in jmem_heap_alloc_block /share/repos/jerryscript/jerry-core/jmem/jmem-heap.c:373
    #4 0x565f80f2 in ecma_alloc_property_pair /share/repos/jerryscript/jerry-core/ecma/base/ecma-alloc.c:198
    #5 0x56675490 in ecma_create_property /share/repos/jerryscript/jerry-core/ecma/base/ecma-helpers.c:457
    #6 0x56675b61 in ecma_create_named_data_property /share/repos/jerryscript/jerry-core/ecma/base/ecma-helpers.c:543
    #7 0x5664a3cf in ecma_op_general_object_define_own_property /share/repos/jerryscript/jerry-core/ecma/operations/ecma-objects-general.c:346
    #8 0x5663cbcd in ecma_op_array_object_define_own_property /share/repos/jerryscript/jerry-core/ecma/operations/ecma-array-object.c:343
    #9 0x5660914b in ecma_op_object_define_own_property /share/repos/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1219
    #10 0x5662c085 in ecma_builtin_helper_def_prop /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-helpers.c:848
    #11 0x5663c20f in ecma_op_create_array_object /share/repos/jerryscript/jerry-core/ecma/operations/ecma-array-object.c:119
    #12 0x5660398c in ecma_builtin_array_dispatch_construct /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array.c:101
    #13 0x56603925 in ecma_builtin_array_dispatch_call /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array.c:87
    #14 0x56635af5 in ecma_builtin_dispatch_call /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1051
    #15 0x56644416 in ecma_op_function_call /share/repos/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:727
    #16 0x566310e2 in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1405
    #17 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #18 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #19 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #20 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #21 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #22 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #23 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #24 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #25 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #26 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #27 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #28 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    [ ... duplicate lines ... ]
    #239 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #240 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #241 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #242 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #243 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #244 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #245 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #246 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #247 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751
    #248 0x5663166b in ecma_builtin_json_str /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1499
    #249 0x56632801 in ecma_builtin_json_array /share/repos/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1751

SUMMARY: AddressSanitizer: stack-overflow (/usr/lib32/libasan.so.4+0xe5ea7) in malloc
==98861==ABORTING
```
When creating a non-ASAN binary with build command:
```
./tools/build.py --clean --debug --compile-flag=-m32 \
                          --profile=es2015-subset --system-allocator=on \
                          --error-messages=on --logging=on
```
Jerry runs for a while then it segfaults:
```
$ ./build/bin/jerry /share/js_validation/test_03.js
Segmentation fault (core dumped)
```
A similar behaviour can be observed when running the test in node:
```
$ node /share/js_validation/test_03.js
/share/js_validation/test_03.js:1
(function (exports, require, module, __filename, __dirname) { JSON.stringify(10, Array)
                                                                   ^

RangeError: Maximum call stack size exceeded
    at JSON.stringify (<anonymous>)
    at Object.<anonymous> (/share/js_validation/test_03.js:1:68)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
    at startup (internal/bootstrap/node.js:279:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:696:3)

$ node --version
v10.10.0
```

rerobika, 2019-07-11:
Note: duplicate of #2258.

rerobika, 2019-07-15:
@dodgerblue Thanks for the report I've submitted a PR to resolve this problem.

dodgerblue, 2019-07-21:
Thank you, @rerobika !

PR #2969 (2019-07-23): Rework the engine's internal recursion limit
This patch unifies the recursion limit checking for RegExp, function call and JSON as well.
Until now the limit was only a counter which was increased/decreased at certain points.
This counter has been substituted with a numeric limit which allows to restrict the stack usage.

This patch fixes #2963 and resolves the closed #2258 issue.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 4a9e1858 (2019-07-23): Rework the engine's internal recursion limit (#2969)

This patch unifies the recursion limit checking for RegExp, function call and JSON as well.
Until now the limit was only a counter which was increased/decreased at certain points.
This counter has been substituted with a numeric limit which allows to restrict the stack usage.

This patch fixes #2963 and resolves the closed #2258 issue.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-07-23 by rerobika