jerryscript-project/jerryscript #3011: How to get error position (row, column) in code?

niklauslee, 2019-08-09:
Hello,

Is there a way to get error position (row, column) in code?

For example:
```
jerry_value t parsed_code = ...;
jerry_value_t result = jerry_run(parsed_code);
if (jerry_value_is_error(result)) {
  // want to know the position where the error was originally raised
}
```

I'm developing JavaScript IDE to support JerryScript. In here, I want to show error position in the Code Editor directly.

Moreover, I'm using Babel and Webpack to support users can use latest ECMAScript standard and use `import`, `require()` for import third-party modules. The transpiled and bundled code is executed by JerryScript. If there was an error, I need to show the error position in the original code (not transpiled/bundled) using `source-map`.

P.S.) Here is the project I'm working: https://kameleon.io

Thanks,

akosthekiss, 2019-08-09:
Hi, you might take a look at `print_unhandled_exception` in jerry-main/main-unix.c . It is not a totally general solution but it might give you a start. (Also note that you'll need `JERRY_ERROR_MESSAGES` enabled at build time to have any meaningful error messages, IIRC.)

niklauslee, 2019-08-10:
@akosthekiss Thank you for answer! The example show how to get error position for syntax error.

But actually I want to know is getting error position for runtime errors with stack backtrace.

For example:
```js
function f1 () {
  f2(); // unknown function call
}
f1();
```

Expected results (__This is Node.js's result__):
```
ReferenceError: f2 is not defined
    at f1 (/Users/niklaus/test.js:2:2)
    at Object.<anonymous> (/Users/niklaus/test.js:4:1)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
```

It is good to get error position of a runtime error with a snapshot of backtrace in the `print_unhandled_exception()`. 

Is there any possible way?

Thank you in advance.

LaszloLango, 2019-08-13:
See also: #2281

dbatyai, 2019-08-13:
Hi @niklauslee,

If you have `JERRY_LINE_INFO` enabled, runtime exceptions will contain a "stack" property, which you can use to get the JS stack trace. As @akosthekiss mentioned, take a look at [main-unix.c:98](https://github.com/jerryscript-project/jerryscript/blob/master/jerry-main/main-unix.c#L98), you can find how to retrieve the stack trace there.

One thing to keep in mind, we do not currently support column info, only line numbers.

niklauslee, 2019-08-13:
Thanks @dbatyai. It works well :)

It would be really nice if column info is supported because bundled (minified) code is executed in our project to reduce code size stored in flash memory in MCU.

As you know, bundled (minified) code has only has a single line by removing all white spaces. To show error position, we are using **source-map** to map from the bundled (minified) code to the original code. Without column info, we cannot map from bundled code to the original code :(

Thank you.

zherczeg, 2020-01-13:
Currently we update the current position as executing special byte codes. Adding column info would greatly increase the performance costs of these updates. An alternative would be to create a source map, but that would require lot more memory. We might do that in the future but no changes are planned now.

Hexxeh, 2020-02-21:
Would a PR that adds column info in the same manner as line info is currently tracked be acceptable, or is the sourcemap approach preferred?

zherczeg, 2020-02-21:
Some kind of compressed line/column info would be the best option. Optionally it would only contain lines (configured by a compile time switch).

rerobika, 2021-12-06:
Now we can obtain the line/column info as well. Closing the issue.

closed 2021-12-06 by rerobika

Hexxeh, 2021-12-06:
Does it work for snapshots? Looking I see a PR from ~5 months ago where it works but not for snapshots.

rerobika, 2021-12-06:
Indeed, snapshots are still unsupported, but the original issue was about only the line/column info. However feel free to open a new issue for snaphot support.