jerryscript-project/jerryscript #3034: How to support builtin modules for ES6 import

niklauslee, 2019-09-02:
Hi,

I have some builtin modules such as `gpio`, `i2c`, `spi`, etc and I want to know how to support ES6's `import from` like below:

```js
import {I2C} from 'i2c'
```

I had read https://jerryscript.net/ext-reference-module/ and tried `JERRYX_NATIVE_MODULE ` like below:

```c
static jerry_value_t i2c_module_on_resolve (void) {
  return ...;
}

JERRYX_NATIVE_MODULE (i2c, i2c_module_on_resolve)

void main() {
  ...
  i2c_register();
  ...
}
```

But `SyntaxError: Requested module not found. [<anonymous>:1:19]` error returned when try to use `import {I2C} from 'i2c'`.

Actually I already have my own module loader `require()`, so when user try to use `import` I want to call `require()` internally.

Any examples, advices or suggestions?

Thanks in advance.

rerobika, 2019-09-02:
cc @dbatyai.

dbatyai, 2019-09-02:
Hi @niklauslee,

Currently the jerry-ext native module implementation is not able to directly interwork with the core module system, because the core won't have access to the modules defined with the ext macros.

Unfortunatley there isn't a specific import handler which could be rewired to call require, as import/export statements are handled by the parser.

What we could do, is provide an API function similar to the `JERRYX_NATIVE_MODULE` macro, which could register a canonical name and a resolver into the core module system, and patch the module resoution to handle them. At a quick glance it shouldn't require any major changes in the current code, and you could reuse the resolvers you already have.

Would this be sufficient for your use case? The downside would be that if you'd like to use both require and import at the same time, then you'd have to register the native module twice.

niklauslee, 2019-09-03:
@dbatyai, thank for your answer.

I understand that ES6 import is not related with jerry-ext native module.

So currently no way to provide builtin modules (or custom modules from file system) for ES6 import? Or, is there some way if I do not use `require()` internally?

If currently no way for that, it would be nice if some ways for registering builtin or custom modules or registering callback custom resolver functions (preferred) for ES6 import.

Registering twice is ok if I can avoid duplicated module caching. (one for my own `require()` and one for JerryScript's internal cache).

Thank you.

bil-ash, 2019-09-05:
> Hi @niklauslee,
>
> Currently the jerry-ext native module implementation is not able to directly interwork with the core module system, because the core won't have access to the modules defined with [...]

As of now (after the release of v2.0),the js APIs(promise,arrow function,symbols,etc) provided by JerryScript is more than enough from the context of IoT applications. What is now required is an API similar to  [that provided by iotjs](https://github.com/jerryscript-project/iotjs/blob/master/docs/devs/Writing-New-Module.mdl) for writing dynamically loadable(N-API style) native-modules for jerryscript.  This will ensure that only the required modules are loaded, thereby saving RAM. I know that it exists in iotjs but iotjs has a number of dependencies(libtuv,httpparser,etc) and is supported in lesser number of platforms as compared to JerryScript. So providing an API for loading dynamically-loadable native-modules will ensure that if I require only **gpio** and  **spi** APIs in a project but do not require any API to access the **filesystem** I can do so easily. 
Most probably this is also what @niklauslee requires.

niklauslee, 2019-09-05:
Hi @bil-ash, If you need `gpio`, `spi`, `i2c` modules, take a look [my project](https://github.com/kameleon-project/kameleon). It do not require any dependencies other than jerryscript. Hope it's helpful for you.

bil-ash, 2019-09-06:
> Hi @bil-ash, If you need `gpio`, `spi`, `i2c` modules, take a look [my project](https://github.com/kameleon-project/kameleon). It do not require any dependencies other than jerryscript. Hope it's helpful for you.

Thanks for pointing out to your repo. Hoping something similar gets added to jerryscript. Actually, I need to run on zephyr os

niklauslee, 2019-09-15:
Thank you @rerobika. I will try and give feedback as soon as possible.

niklauslee, 2019-09-16:
Hi @rerobika,

I'm trying the pull-request #3090.

I have `require()` function to load builtin modules and try to call `require()` inside `jerry_port_get_native_module`. Below is my `jerry_port.c` implementation.

```c
size_t jerry_port_normalize_path (const char *in_path_p, char *out_buf_p,
    size_t out_buf_size, char *base_file_p) {
  size_t len = strlen (in_path_p);
  if (len + 1 > out_buf_size) {
    return 0;
  }
  strcpy (out_buf_p, in_path_p);
  return len;    
}

jerry_value_t jerry_port_get_native_module (jerry_value_t name) {
  jerry_value_t global = jerry_get_global_object();
  jerry_value_t prop = jerry_create_string((const jerry_char_t *) "require");
  jerry_value_t require_fn = jerry_get_property (global, prop);
  jerry_release_value (prop);

  jerry_value_t args[] = { name };
  jerry_value_t ret_val = jerry_call_function (require_fn, global, args, 1);

  jerry_release_value(require_fn);
  jerry_release_value(global);

  return ret_val;
}
```

And then I tried to parse and run the below:

```js
import {GPIO} from 'gpio';
```

However, the system gets stuck.

And, I just have realized that `jerry_port_get_native_module` is called during `jerry_parse()`, not `jerry_run()`. I think I'm not correctly understanding how `import` statement is processed.

So my questions are:

1. This kind of usage (calling `require()` in `jerry_port_get_native_module`) is possible?
2. If not, what conditions are required for implementing `jerry_port_get_native_module`? It should return only a parsed javascript source?

> **For more info:** `require()` is implemented by JavaScript and complied into a snapshot. The snapshot is executed using `jerry_exec_snapshot()` at system startup. 

Thank you in advance.

dbatyai, 2019-09-16:
Hi @niklauslee,

`require` and `import` work fundamentally differently. `require` is used to load modules dynamically during script evaluation, while `import` will load modules statically, before the current parsed source is evaluated. One other key difference is that `require` will always load the whole module object, while an `import` statement is able to load only the specified variables.

What this means is calling a js function during an import statement will result in an undefined behavior, as the source has not yet been evaluated, and the variable assignments did not yet happen.

The intention of the `jerry_port_get_native_module` functions is for embedders to be able to provide pure native modules to the engine. The implementation requirements are the following:
a) The `jerry_port_get_native_module` function should return a 'namespace' object that contains all the exported variables for the specific module. (roughly equivalent to module.exports)
b) The namespace object should be created using only native code and the Jerry C API.

For example in your implementation, the I2C module would be like this. When `jerry_port_get_native_module` is called with the `"i2c"` argument, you could return the object returned by your `module_i2c_init` function. The module will be cached in the engine, so the function will only get called once.

For modules that are implemented in Javascript, I'd suggest to use the export statement.
For example in your `GPIO` module, you could write `export {GPIO};`, and then the `import {GPIO} from 'gpio';` statement should work, as long as the `'gpio'` module specifier resolves to the path of your gpio.js module source file. This could be achieved using the `jerry_port_normalize_path` function. When this function is called with the `gpio` zero terminated string as the input path, you should return a (preferably absolute) path to your gpio.js module file. Returned relative paths will be resolved from where the engine was started, you may or may not be ably to use this to your advantage.

So unfortunately calling your `require` implementation during an import will not be possible due to the differences mentioned above. The two possible ways to achieve a similar result is to use pure native modules with `jerry_port_get_native_module` function, or use Javascript modules with export statements.

One other thing to note, since you mentioned snapshots. Import/export statements are not yet supported with snapshots, so depending on how you use snapshots, it may not work outright, or have unexpected results.

niklauslee, 2019-09-16:
@dbatyai Thank you for fast response. Now I understand the different between `require` and `import`.

Our module loading system should be re-written entirely to support `import` statement. So we decided to delay supporting this feature in the future.

Thank you all for your supports 👍

PR #3090 (2019-10-02): Provide support for native modules with ES6 imports
Resolves #3034.

JerryScript-DCO-1.0-Signed-off-by: Dániel Bátyai dbatyai@inf.u-szeged.hu
closed 2019-10-02 by dbatyai