jerryscript-project/jerryscript #3057: Assertion uint_ptr % JMEM_ALIGNMENT == 0 in jmem_decompress_pointer

renatahodovan, 2019-09-04:
###### JerryScript revision
1088273b

###### Build platform
Linux-4.15.0-54-generic-x86_64-with-Ubuntu-18.04-bionic

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
var b  = [""];
function dConstr ( ) { }
dConstr.prototype = b;
var d =   new dConstr;
for (var $ in  d);
```

###### Output
```text
ICE: Assertion 'uint_ptr % JMEM_ALIGNMENT == 0' failed at jerryscript/jerry-core/jmem/jmem-allocator.c(jmem_decompress_pointer):222.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

###### Backtrace
```text
bt
#0  0xf7fd5059 in __kernel_vsyscall ()
#1  0xf77fc832 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf77fdcc1 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0x5657ae47 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-port/default/default-fatal.c:71
#4  0x56617934 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x56617975 in jerry_assert_fail (assertion=0x566ab760 "uint_ptr % JMEM_ALIGNMENT == 0", file=0x566ab640 "jerryscript/jerry-core/jmem/jmem-allocator.c", function=0x5667ef80 <__func__.4606.lto_priv.506> "jmem_decompress_pointer", line=222) at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x56616ade in jmem_decompress_pointer (compressed_pointer=4126148881) at jerryscript/jerry-core/jmem/jmem-allocator.c:222
#7  0x56607c91 in ecma_op_object_get_property_names (obj_p=0xf5d005f0, opts=6) at jerryscript/jerry-core/ecma/operations/ecma-objects.c:1829
#8  0x565757e4 in opfunc_for_in (left_value=4124050931, result_obj_p=0xffffc680) at jerryscript/jerry-core/vm/opcodes.c:248
#9  0x565f4c00 in vm_loop (frame_ctx_p=0xffffc7e0) at jerryscript/jerry-core/vm/vm.c:2840
#10 0x565f9958 in vm_execute (frame_ctx_p=0xffffc7e0, arg_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:3616
#11 0x565fa2bc in vm_run (bytecode_header_p=0xf51031d0, this_binding_value=4126149459, lex_env_p=0xf5d007b0, parse_opts=0, arg_list_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:3742
#12 0x565e7a1c in vm_run_global (bytecode_p=0xf51031d0) at jerryscript/jerry-core/vm/vm.c:282
#13 0x5664170c in jerry_run (func_val=4126148451) at jerryscript/jerry-core/api/jerry.c:570
#14 0x5663e070 in main (argc=3, argv=0xffffcc14) at jerryscript/jerry-main/main-unix.c:743
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

rerobika, 2019-09-04:
Related to #3050, #3053 will resolve this issue.

rerobika, 2019-09-05:
#3053 Landed, closing this issue.

closed 2019-09-05 by rerobika