jerryscript-project/jerryscript #3058: Assertion ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)

made0x78, 2019-09-04:
###### Revision 
1088273

###### Build
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer
--compile-flag=-fno-common --compile-flag=-g
--strip=off --system-allocator=on --logging=on
--error-messages=on --profile=es2015-subset
```

###### OS 
Linux 4.15.0-58-generic #64-Ubuntu x86_64 GNU/Linux

###### Test case
```javascript
var arr = [1];
var t = arr.find(function(a) {
return Symbol.for(null);
});
```

###### Backtrace
Run with `jerry --abort-on-fail poc.js`

```
ICE: Assertion 'ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)' failed at /jerryscript/jerry-core/ecma/base/ecma-literal-storage.c(ecma_free_symbol_list):46.
Error: ERR_FAILED_INTERNAL_ASSERTION

(gdb) bt
#0  0xf7fd5059 in __kernel_vsyscall ()
#1  0xf7823452 in raise () from /lib32/libc.so.6
#2  0xf7824871 in abort () from /lib32/libc.so.6
#3  0x5657ae47 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at /jerryscript/jerry-port/default/default-fatal.c:71
#4  0x56617934 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at /jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x56617975 in jerry_assert_fail (assertion=0x566b8bc0 "ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)", 
    file=0x566b8b60 "/jerryscript/jerry-core/ecma/base/ecma-literal-storage.c", function=0x5667a300 <__func__.5128.lto_priv.49> "ecma_free_symbol_list", line=46)
    at /jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x5666c9bc in ecma_free_symbol_list (symbol_list_cp=4124051248) at /jerryscript/jerry-core/ecma/base/ecma-literal-storage.c:46
#7  0x5666cd32 in ecma_finalize_lit_storage () at /jerryscript/jerry-core/ecma/base/ecma-literal-storage.c:119
#8  0x5666b979 in ecma_finalize () at /jerryscript/jerry-core/ecma/base/ecma-init-finalize.c:64
#9  0x56640889 in jerry_cleanup () at /jerryscript/jerry-core/api/jerry.c:216
#10 0x5663e896 in main (argc=3, argv=0xffffd3d4) at /jerryscript/jerry-main/main-unix.c:941
```

rerobika, 2019-09-04:
Thanks for the report! This issue is related to #3049, and #3052 will resolve it.

rerobika, 2019-09-05:
#3052 Landed, closing the issue.

closed 2019-09-05 by rerobika