jerryscript-project/jerryscript #3059: SEGV  in ecma_ref_ecma_string

renatahodovan, 2019-09-04:
###### JerryScript revision
1088273b

###### Build platform
Linux-4.15.0-54-generic-x86_64-with-Ubuntu-18.04-bionic

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
var b = [$, ""];
function dConstr ( ) { }
dConstr.prototype = b;
var d = new dConstr;
for (var $ in d);
```

###### Output
```text
ASAN:DEADLYSIGNAL
=================================================================
==49145==ERROR: AddressSanitizer: SEGV on unknown address 0x00000068 (pc 0x5668a412 bp 0xff944168 sp 0xff944150 T0)
==49145==The signal is caused by a READ memory access.
==49145==Hint: address points to the zero page.
    #0 0x5668a411 in ecma_ref_ecma_string jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:825
    #1 0x5668d5ab in ecma_string_from_property_name jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:1632
    #2 0x56636ed5 in ecma_op_object_get_property_names jerryscript/jerry-core/ecma/operations/ecma-objects.c:1849
    #3 0x565a47e3 in opfunc_for_in jerryscript/jerry-core/vm/opcodes.c:248
    #4 0x56623bff in vm_loop jerryscript/jerry-core/vm/vm.c:2840
    #5 0x56628957 in vm_execute jerryscript/jerry-core/vm/vm.c:3616
    #6 0x566292bb in vm_run jerryscript/jerry-core/vm/vm.c:3742
    #7 0x56616a1b in vm_run_global jerryscript/jerry-core/vm/vm.c:282
    #8 0x5667070b in jerry_run jerryscript/jerry-core/api/jerry.c:570
    #9 0x5666d06f in main jerryscript/jerry-main/main-unix.c:743
    #10 0xf7788e80 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18e80)
    #11 0x5659a6e0  (jerryscript/build_gcc_asan_es2015/bin/jerry+0x166e0)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:825 in ecma_ref_ecma_string
==49145==ABORTING
```

###### Backtrace
```text
bt
#0  0x5665b412 in ecma_ref_ecma_string (string_p=0x68) at jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:825
#1  0x5665e5ac in ecma_string_from_property_name (property=9 '\t', prop_name_cp=104) at jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:1632
#2  0x56607ed6 in ecma_op_object_get_property_names (obj_p=0xf5d00650, opts=6) at jerryscript/jerry-core/ecma/operations/ecma-objects.c:1849
#3  0x565757e4 in opfunc_for_in (left_value=4124051027, result_obj_p=0xffffc680) at jerryscript/jerry-core/vm/opcodes.c:248
#4  0x565f4c00 in vm_loop (frame_ctx_p=0xffffc7e0) at jerryscript/jerry-core/vm/vm.c:2840
#5  0x565f9958 in vm_execute (frame_ctx_p=0xffffc7e0, arg_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:3616
#6  0x565fa2bc in vm_run (bytecode_header_p=0xf5101ea0, this_binding_value=4126149459, lex_env_p=0xf5d007b0, parse_opts=0, arg_list_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:3742
#7  0x565e7a1c in vm_run_global (bytecode_p=0xf5101ea0) at jerryscript/jerry-core/vm/vm.c:282
#8  0x5664170c in jerry_run (func_val=4126148835) at jerryscript/jerry-core/api/jerry.c:570
#9  0x5663e070 in main (argc=3, argv=0xffffcc14) at jerryscript/jerry-main/main-unix.c:743
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with  [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

rerobika, 2019-09-04:
Related to #3050, #3053 will resolve this issue.

rerobika, 2019-09-05:
#3053 landed, closing this issue.

closed 2019-09-05 by rerobika