jerryscript-project/jerryscript #3071: Assertion 'id < LIT_NON_INTERNAL_MAGIC_STRING__COUNT' in lit-magic-strings.c

made0x78, 2019-09-05:
###### Revision
57f389d

###### Build
./tools/build.py --clean --debug --compile-flag=-fsanitize=address
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer
--compile-flag=-fno-common --compile-flag=-g
--strip=off --system-allocator=on --logging=on
--error-messages=on --profile=es2015-subset

###### OS
Linux 4.15.0-58-generic #64-Ubuntu x86_64 GNU/Linux

###### Test case
```javascript
var set = new Set();
var arrb = new ArrayBuffer(1);
var arr8 = new Uint8Array(arrb, 0);
delete set.add(arr8);
```

###### Backtrace
Run with `jerry --abort-on-fail poc.js`

```
ICE: Assertion 'id < LIT_NON_INTERNAL_MAGIC_STRING__COUNT' failed at /jerryscript/jerry-core/lit/lit-magic-strings.c(lit_get_magic_string_size):82.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0xf7fd5059 in __kernel_vsyscall ()
(gdb) bt
#0  0xf7fd5059 in __kernel_vsyscall ()
#1  0xf7841452 in raise () from /lib32/libc.so.6
#2  0xf7842871 in abort () from /lib32/libc.so.6
#3  0x566bdfef in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at /jerryscript/jerry-port/default/default-fatal.c:71
#4  0x566513cf in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at /jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x56651410 in jerry_assert_fail (assertion=0x566e5f20 "id < LIT_NON_INTERNAL_MAGIC_STRING__COUNT", file=0x566e5ec0 "/jerryscript/jerry-core/lit/lit-magic-strings.c",
    function=0x567029a0 <__func__.18616> "lit_get_magic_string_size", line=82) at /jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x56653132 in lit_get_magic_string_size (id=319) at /jerryscript/jerry-core/lit/lit-magic-strings.c:82
#7  0x5658e02e in ecma_string_get_chars_fast (string_p=0x27ed, size_p=0xffffc740) at /jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:147
#8  0x5659369e in ecma_string_to_number (string_p=0x27ed) at /jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:951
#9  0x5662f394 in ecma_op_object_get_own_property (object_p=0xf5f006d0, property_name_p=0x27ed, property_ref_p=0xffffca20, options=0)
    at /jerryscript/jerry-core/ecma/operations/ecma-objects.c:203
#10 0x5661585b in ecma_op_container_to_key (key_arg=4126148307) at /jerryscript/jerry-core/ecma/operations/ecma-container-object.c:250
#11 0x56616285 in ecma_op_container_set (this_arg=4126148739, key_arg=4126148307, value_arg=4126148307, lit_id=LIT_MAGIC_STRING_SET_UL)
    at /jerryscript/jerry-core/ecma/operations/ecma-container-object.c:403
#12 0x565ee83d in ecma_builtin_set_prototype_object_add (this_arg=4126148739, value_arg=4126148307) at /jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-set-prototype.c:50
#13 0x565ee66a in ecma_builtin_set_prototype_dispatch_routine (builtin_routine_id=71, this_arg_value=4126148739, arguments_list=0xffffcba0, arguments_number=1)
    at /jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-set-prototype.inc.h:41
#14 0x565ffbe2 in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_SET_PROTOTYPE, builtin_routine_id=71, this_arg_value=4126148739, arguments_list_p=0xffffcba0,
    arguments_list_len=1) at /jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1025
#15 0x565ffe43 in ecma_builtin_dispatch_call (obj_p=0xf5f006a0, this_arg_value=4126148739, arguments_list_p=0xffffcefc, arguments_list_len=1)
    at /jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1050
#16 0x56621b54 in ecma_op_function_call (func_obj_p=0xf5f006a0, this_arg_value=4126148739, arguments_list_p=0xffffcefc, arguments_list_len=1)
    at /jerryscript/jerry-core/ecma/operations/ecma-function-object.c:729
#17 0x566a6be0 in opfunc_call (frame_ctx_p=0xffffcf90) at /jerryscript/jerry-core/vm/vm.c:581
#18 0x566bbd84 in vm_execute (frame_ctx_p=0xffffcf90, arg_p=0x0, arg_list_len=0) at /jerryscript/jerry-core/vm/vm.c:3622
#19 0x566bc633 in vm_run (bytecode_header_p=0xf5301ad0, this_binding_value=4126149459, lex_env_p=0xf5d007b0, parse_opts=0, arg_list_p=0x0, arg_list_len=0)
    at /jerryscript/jerry-core/vm/vm.c:3742
#20 0x566a5a46 in vm_run_global (bytecode_p=0xf5301ad0) at /jerryscript/jerry-core/vm/vm.c:282
#21 0x56570a9c in jerry_run (func_val=4126148979) at /jerryscript/jerry-core/api/jerry.c:570
#22 0x5656d3ef in main (argc=3, argv=0xffffd3f4) at /jerryscript/jerry-main/main-unix.c:743
```

rerobika, 2019-09-05:
This issue is related to #3062. #3065 will resolve this problem.

rerobika, 2019-09-09:
Closing via #3065.

closed 2019-09-09 by rerobika