jerryscript-project/jerryscript #3142: error adding symbols: Bad value

cup, 2019-09-21:
So first I was getting this:

~~~
$ make
[ 10%] Built target jerry-libm
[ 85%] Built target jerry-core
[ 93%] Built target jerry-ext
[ 96%] Built target jerry-port-default-minimal
[ 99%] Built target jerry-port-default
[100%] Linking C executable ../bin/jerry.exe
/bin/ld: unrecognized option '-z'
~~~

So I removed this:

https://github.com/jerryscript-project/jerryscript/blob/25b81c106c07f33f576fb24c5c636d5799ac1eff/CMakeLists.txt#L183

but it still fails:

~~~
$ make
[ 10%] Built target jerry-libm
[ 85%] Built target jerry-core
[ 93%] Built target jerry-ext
[ 96%] Built target jerry-port-default-minimal
[ 99%] Built target jerry-port-default
[100%] Linking C executable ../bin/jerry.exe
/bin/ld: ../lib/libjerry-core.a(ecma-helpers-number.c.o): bad string table size 0
../lib/libjerry-core.a: error adding symbols: Bad value
~~~

akosthekiss, 2019-09-22:
- What is your build platform? Architecture, OS, version?
- What is your compiler, linker? Name, version?
- What is your build target? Architecture, OS, version (if applicable)

cup, 2019-09-22:
~~~
$ uname -a
CYGWIN_NT-6.1 Steven-PC 3.0.7(0.338/5/3) 2019-04-30 18:08 x86_64 Cygwin

$ find /bin -samefile /bin/gcc.exe
/bin/gcc.exe
/bin/x86_64-pc-cygwin-gcc-7.4.0.exe
/bin/x86_64-pc-cygwin-gcc.exe
~~~

akosthekiss, 2019-09-22:
@cup Please, note that cygwin has never been among the supported platforms of this project. Anyway, I cc: @ossy-szeged , who has been working on the Windows support lately. Perhaps he knows what's going on there. But you are also welcome to contribute if you can find the fix to the build system.

ossy-szeged, 2019-09-23:
I tried cygwin build and got the following error message:

collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/bin/ld: ../lib/libjerry-core.a(ecma-builtin-json.c.o) (/711
/10): Section flag STYP_GROUP (0x4) ignored
/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/bin/ld: ../lib/libjerry-core.a(ecma-builtin-json.c.o) (/711
/10): Section flag STYP_COPY (0x10) ignored
/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/bin/ld: ../lib/libjerry-core.a(ecma-builtin-json.c.o): warning: COMDAT symbol '.gnu.lto_ecma_builtin_json_parse_next_token.6ef7c7fa669409d5' does not match section name '/711
/10'

And I found this related cygwin bug report:
http://cygwin.1069669.n5.nabble.com/Error-Linking-LTO-programs-using-GCC6-4-td136659.html

It seems LTO is still buggy on cygwin (tried with latest cygwin with GCC 7.4). But without LTO enabled, the build suceeded for me. Until this bug is fixed in cygwin, you should pass --lto=OFF to build.py.

Otherwise do you have any reason to use cygwin instead of native Windows build? The native build works fine and tested day by day. You only need cmake, python and MSVC installed.

closed 2019-09-23 by rerobika

cup, 2019-09-23:
> Otherwise do you have any reason to use cygwin instead of native Windows
> build? The native build works fine and tested day by day. You only need cmake,
> python and MSVC installed.

Using the current version with the pull request merged, solved the issue for me.

In regard to Cygwin version vs Native Windows build, I dont have a preference,
I just figured the Cygwin build would be easier to make. Also I didnt realize
that builds are already available here:

https://ci.appveyor.com/project/jerryscript-project/jerryscript

that should probably be mentioned somewhere, so people are building for no
reason. In regard to building the native version, by MSVC I assume you mean
Visual Studio? If I am building native I will use mingw64, as its about 2 orders
of magnitude smaller than Visual Studio.

At any rate, thanks all.

cup, 2019-09-23:
https://github.com/jerryscript-project/jerryscript/pull/3147

galpeter, 2019-09-24:
A few things to point out:
* Please note that the CI builds are for testing.
* The JerryScript is (mainly) library with lots of build/runtime options. It is possible that the options defined for CI build/testing is are too much/few for a given project/usage. There are also no pre-built shared libraries for other platforms.
* The jerryscript-main (from which the executable is built) is an example REPL for the project.
* Regarding the binary sizes it is interesting to hear, it would be good to have concrete numbers (in a different issue). What I'm guessing there is a difference on compiler options.

PR #3147 (2019-09-23): Fix Cygwin build
Fixes #3142.

Changes:
- Disable LTO on Cygwin build because of a Cygwin-ld bug.
- Don't add "-z noexecstack" to ld because it is ELF specific option.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác oszi@inf.u-szeged.hu
commit 053b05f1 (2019-09-23): Fix Cygwin build (#3147)

Fixes #3142.

Changes:
- Disable LTO on Cygwin build because of a Cygwin-ld bug.
- Don't add "-z noexecstack" to ld because it is ELF specific option.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác oszi@inf.u-szeged.hu