jerryscript-project/jerryscript #3144: Help with getting name of registered jerry_external_handler_t name, please.

rafail-ufa, 2019-09-21:
How to get information from const **jerry_value_t function_object** and **const jerry_value_t function_this** in code like below. I need to get the name of called external handler (some_handler) inside this handler. 
Please help. And what is the structure of **function_object** and **function_this** to use them in a code.

`#include <stdio.h>
#include "jerryscript.h"

static jerry_value_t
some_handler (const jerry_value_t function_object,
               const jerry_value_t function_this,
               const jerry_value_t arguments[],
               const jerry_length_t arguments_count)
{
	
  return jerry_create_undefined ();
}

int
main (void)
{
  const jerry_char_t script[] = "call_handler('First arg');";
  const jerry_length_t script_size = sizeof (script) - 1;

  jerry_init (JERRY_INIT_EMPTY);

  {
    jerry_value_t global_object = jerry_get_global_object ();
    jerry_value_t property_name_print = jerry_create_string ((const jerry_char_t *) "call_handler");
    jerry_value_t property_value_func = jerry_create_external_function (print_handler);
    jerry_value_t set_result = jerry_set_property (global_object, property_name_print, property_value_func);

    jerry_release_value (set_result);
    jerry_release_value (property_value_func);
    jerry_release_value (property_name_print);
    jerry_release_value (global_object);
  }

  jerry_value_t parsed_code = jerry_parse (NULL, 0, script, script_size, JERRY_PARSE_NO_OPTS);

  if (!jerry_value_is_error (parsed_code))
  {
    jerry_value_t ret_value = jerry_run (parsed_code);
    jerry_release_value (ret_value);
  }
  jerry_release_value (parsed_code);
  jerry_cleanup ();
  return 0;
}`

Thanks a lot!

galpeter, 2019-09-26:
Hi!

By default an external method (created via `jerry_create_external_function`) does not have a name.

However, you can attach a `name` property to the created function object. This `name` property then can be accessed via the `function_object` argument in the C method (or at any other place if required).

Example to set/get `name` property for an external function (snippet):

```c
static jerry_value_t some_handler (
    const jerry_value_t function_object,
    const jerry_value_t function_this,
    const jerry_value_t arguments[],
    const jerry_length_t arguments_count)
{   
    jerry_value_t name_val = jerry_create_string ((const jerry_char_t *) "name");
    jerry_value_t prop = jerry_get_property (function_object, name_val);

    jerry_size_t string_size = jerry_get_string_size (prop);
    jerry_char_t prop_str[string_size + 1];

    jerry_size_t copied =
        jerry_substring_to_char_buffer (prop,
                                        0,
                                        string_size,
                                        prop_str,
                                        string_size);
    prop_str[copied] = '\0';

    jerry_release_value (prop);
    jerry_release_value (name_val);

    printf("Called function with name: %s\n", prop_str);

    return jerry_create_undefined;
}

void set_function_name(jerry_value_t func, char* name) {
    jerry_value_t name_key = jerry_create_string ((const jerry_char_t *) "name");
    jerry_value_t name_value = jerry_create_string ((const jerry_char_t *) name);
    jerry_value_t set_result = jerry_set_property (func, name_key, name_value);
    // TODO: check set result if there is an error
    jerry_release_value(set_result);
    jerry_release_value(name_value);
    jerry_release_value(name_key);
}

int main() {
    const jerry_char_t script[] = "call_handler('First arg');";
    const jerry_length_t script_size = sizeof (script) - 1;

    current_instance_p = jerry_create_instance(512 * 1024, instance_alloc, NULL);
    jerry_init (JERRY_INIT_EMPTY);

    {   
        jerry_value_t global_object = jerry_get_global_object ();
        jerry_value_t property_name_print =
            jerry_create_string ((const jerry_char_t *) "call_handler");
        jerry_value_t property_value_func = jerry_create_external_function (some_handler);

        set_function_name (property_value_func, "MyFunction");

        jerry_value_t set_result =
            jerry_set_property (global_object, property_name_print, property_value_func);
        // TODO: check the 'set_result' if it is an error.
        jerry_release_value (set_result);
        jerry_release_value (property_value_func);
        jerry_release_value (property_name_print);
        jerry_release_value (global_object);

    }

    jerry_value_t parsed_code =
        jerry_parse (NULL, 0, script, script_size, JERRY_PARSE_NO_OPTS);

    if (!jerry_value_is_error (parsed_code))
    {   
        jerry_value_t ret_value = jerry_run (parsed_code);
        jerry_release_value (ret_value);
    }

    jerry_release_value (parsed_code);
    jerry_cleanup ();
    return 0;
}
```

In case of a native C method the `function_object` argument means the function object which was invoked (in this case this was crated via the `jerry_create_external_function` method). The `function_this` is the `this` object which used to invoke the method.

For example if the following JS code is executed in this example:

```js
call_handler("First arg");
```

The `function_this` value is an `undefined` value (as there was no 'this' set for the function). 

However, if the following JS code is executed:

```js
var my_demo = { val: 6 }
call_handler.call(my_demo, "First arg");
```

The `function_this` will be the object referenced by the `my_demo` var.

rafail-ufa, 2019-09-26:
Hi! 
That really works.
I thought that can get function name from function object, but really, if there is no property you can add your own.

 Thank you very much!

> Hi!
>
> By default an external method (created via `jerry_create_external_function`) does not have a name.
>
> However, you can attach a `name` property to the created function object. This `name` [...]

closed 2019-09-27 by rafail-ufa