jerryscript-project/jerryscript #3151: Assertion context_p->token.type != LEXER_RIGHT_PAREN in parser_parse_while_statement_start

renatahodovan, 2019-09-23:
###### JerryScript revision
35a3df3f

###### Build platform
Linux-4.15.0-62-generic-x86_64-with-Ubuntu-18.04-bionic

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
do function $ ( ) { } while ($);
```

###### Output
```text
ICE: Assertion 'context_p->token.type != LEXER_RIGHT_PAREN' failed at jerryscript/jerry-core/parser/js/js-parser-statm.c(parser_parse_while_statement_start):761.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

###### Backtrace
```text
bt
#0  0xf7fd5079 in __kernel_vsyscall ()
#1  0xf77fc832 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf77fdcc1 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0x565798c3 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-port/default/default-fatal.c:71
#4  0x5661550e in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x5661554f in jerry_assert_fail (assertion=0x566a6780 "context_p->token.type != LEXER_RIGHT_PAREN", file=0x566a60c0 "jerryscript/jerry-core/parser/js/js-parser-statm.c", function=0x56696700 <__func__.5348.lto_priv.191> "parser_parse_while_statement_start", line=761) at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x565ae79f in parser_parse_while_statement_start.lto_priv.183 (context_p=0xffffc4a0) at jerryscript/jerry-core/parser/js/js-parser-statm.c:761
#7  0x5656fd8c in parser_parse_statements (context_p=0xffffc4a0) at jerryscript/jerry-core/parser/js/js-parser-statm.c:2397
#8  0x565d18de in parser_parse_source (arg_list_p=0x0, arg_list_size=0, source_p=0x566f0be0 <buffer.lto_priv> "var i = 10 ; \nvar cnt = 0 ; \ndo function Animal ( name ) { this .name = name \nthis .canWalk = true \n} \nwhile ( i -- > 0 ) ; \nprint ( cnt === 0 ) ; \n ", source_size=149, parse_opts=0, error_location_p=0xffffc6b0) at jerryscript/jerry-core/parser/js/js-parser.c:2534
#9  0x565d46c5 in parser_parse_script (arg_list_p=0x0, arg_list_size=0, source_p=0x566f0be0 <buffer.lto_priv> "var i = 10 ; \nvar cnt = 0 ; \ndo function Animal ( name ) { this .name = name \nthis .canWalk = true \n} \nwhile ( i -- > 0 ) ; \nprint ( cnt === 0 ) ; \n ", source_size=149, parse_opts=0, bytecode_data_p=0xffffc770) at jerryscript/jerry-core/parser/js/js-parser.c:3001
#10 0x56649319 in jerry_parse (resource_name_p=0xffffcd8d "/home/reni/.fuzzinator_21989//jerryscript/picireny/138803955492962979049107273345816075348.js", resource_name_length=93, source_p=0x566f0be0 <buffer.lto_priv> "var i = 10 ; \nvar cnt = 0 ; \ndo function Animal ( name ) { this .name = name \nthis .canWalk = true \n} \nwhile ( i -- > 0 ) ; \nprint ( cnt === 0 ) ; \n ", source_size=149, parse_opts=0) at jerryscript/jerry-core/api/jerry.c:426
#11 0x56646109 in main (argc=3, argv=0xffffcb04) at jerryscript/jerry-main/main-unix.c:734
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

zherczeg, 2019-09-24:
This is a known issue and you can do it with a class statement. The best way to fix this would require a goto, but nobody likes goto.

zherczeg, 2019-09-24:
(Btw this is not a prescanner issue, probably there for a long time, but we discovered it recently. Funny thing is it should throw syntax error in newer engines to make this worse.)

akosthekiss, 2019-09-24:
@zherczeg Is this issue report a duplicate? We should close this one then using GitHub's duplicate marking mechanism.

zherczeg, 2019-09-24:
It is not a duplicate. It is a known bug. We discussed it before, but we don't like any solution. We hope somebody will figure out a nice solution for it (no goto, no local variable).

akosthekiss, 2019-09-24:
Ohm, known and discussed by whom and where? Could you link the PR where the alternatives have been discussed/reviewed?

PR #3179 (2019-10-02): Fix class/function parsing in single-statement blocks
This patch fixes #3151.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 32962f06 (2019-10-02): Fix class/function parsing in single-statement blocks (#3179)

This patch fixes #3151 and fixes #3174.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-10-02 by rerobika