jerryscript-project/jerryscript #3161: SEGV  in ecma_get_typedarray_element

renatahodovan, 2019-09-25:
###### JerryScript revision
9ab48722

###### Build platform
Linux-5.0.0-27-generic-x86_64-with-Ubuntu-19.04-disco

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
v4 = Date.UTC(894, 7407, 8094)
var v16 = Int16Array.from(((this)).toString())
v16.lastIndexOf(v4.toExponential.length, v4)
```

###### Output
```text
AddressSanitizer:DEADLYSIGNAL
=================================================================
==22130==ERROR: AddressSanitizer: SEGV on unknown address 0x02f0c25a (pc 0x56619109 bp 0xffc99a48 sp 0xffc99a20 T0)
==22130==The signal is caused by a READ memory access.
    #0 0x56619108 in ecma_get_typedarray_element jerryscript/jerry-core/ecma/operations/ecma-typedarray-object.c:70
    #1 0x565f32ba in ecma_builtin_typedarray_prototype_index_helper jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.c:1755
    #2 0x565f3410 in ecma_builtin_typedarray_prototype_last_index_of jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.c:1797
    #3 0x565ee47b in ecma_builtin_typedarray_prototype_dispatch_routine jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.inc.h:74
    #4 0x565ececd in ecma_builtin_dispatch_routine jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1021
    #5 0x565ed12b in ecma_builtin_dispatch_call jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1046
    #6 0x56600840 in ecma_op_function_call jerryscript/jerry-core/ecma/operations/ecma-function-object.c:729
    #7 0x56641dd0 in opfunc_call jerryscript/jerry-core/vm/vm.c:581
    #8 0x56652f56 in vm_execute jerryscript/jerry-core/vm/vm.c:3636
    #9 0x566537fa in vm_run jerryscript/jerry-core/vm/vm.c:3756
    #10 0x566410d5 in vm_run_global jerryscript/jerry-core/vm/vm.c:282
    #11 0x565b5607 in jerry_run jerryscript/jerry-core/api/jerry.c:576
    #12 0x565b20e6 in main jerryscript/jerry-main/main-unix.c:743
    #13 0xf76bf750 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x1e750)
    #14 0x565af8e0 in _start (jerryscript/build/bin/jerry+0x158e0)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV jerryscript/jerry-core/ecma/operations/ecma-typedarray-object.c:70 in ecma_get_typedarray_element
==22130==ABORTING
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [JsProFuzz](https://www.researchgate.net/publication/303482425_Fuzzing_JavaScript_Engine_APIs). </sup>

szilagyiadam, 2019-09-25:
#3156 Will fix this problem too.

rerobika, 2019-10-07:
Fixed via #3156.

closed 2019-10-07 by rerobika