jerryscript-project/jerryscript #3166: Global out of bounds read of "buffer" array

fumfel, 2019-09-26:
###### JerryScript revision
HEAD: 9ab48722445c483a2b984fef609d13ab87894dea

###### Build platform
`Ubuntu 18.04.3 LTS (Linux 4.15.0-64-generic x86_64) + Clang 9`

###### Build steps

```sh
env CC=clang python tools/build.py --libfuzzer=on --compile-flag=-fsanitize=address --lto=off
```

###### Execution steps

- Download [crashing testcase](https://github.com/jerryscript-project/jerryscript/files/3657991/jerry_goobr_main-unix.txt)
- Run `cat jerry_goobr_main-unix | ./jerry`

###### Output
```
==17233==ERROR: AddressSanitizer: global-buffer-overflow on address 0x000001181a00 at pc 0x0000004c3c3a bp 0x7fff03f8d190 sp 0x7fff03f8d188
READ of size 1 at 0x000001181a00 thread T0
    #0 0x4c3c39  (jerryscript/build/bin/jerry+0x4c3c39)
    #1 0x7feae68c5b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #2 0x41afd9  (jerryscript/build/bin/jerry+0x41afd9)

0x000001181a00 is located 0 bytes to the right of global variable 'buffer' defined in 'jerryscript/jerry-main/main-unix.c:50:16' (0x1081a00) of size 1048576
SUMMARY: AddressSanitizer: global-buffer-overflow (jerryscript/build/bin/jerry+0x4c3c39) 
Shadow bytes around the buggy address:
  0x0000802282f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x000080228300: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x000080228310: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x000080228320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x000080228330: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x000080228340:[f9]f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x000080228350: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x000080228360: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x000080228370: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x000080228380: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x000080228390: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==17233==ABORTING
```

galpeter, 2019-10-02:
Please note that this bug is only related to the example REPL (main-unix.c) and it does not affect the library itself.

rerobika, 2019-10-03:
Fixed via #3193.

closed 2019-10-03 by rerobika