jerryscript-project/jerryscript #3169: Set type

cup, 2019-09-26:
Another issue says that "Set" has been implemented, but it hasnt:

~~~
$ ./jerry
jerry> Set
Script Error: ReferenceError
jerry> Set()
Script Error: ReferenceError
jerry> new Set()
Script Error: ReferenceError
~~~

cup, 2019-09-26:
https://github.com/jerryscript-project/jerryscript/issues/2745

akosthekiss, 2019-09-26:
How did you build jerry? (Command line options to build.py or cmake?)

cup, 2019-09-26:
https://ci.appveyor.com/project/jerryscript-project/jerryscript

akosthekiss, 2019-09-26:
Then you have certainly seen the line in the logs: "JERRY_PROFILE                  es5.1". You'll have to configure and build the project for your own needs (es2015-subset profile, as it seems in this case). Closing as invalid.

closed 2019-09-26 by akosthekiss

cup, 2019-09-26:
~~~
./tools/build.py --profile=es2015-subset
~~~