jerryscript-project/jerryscript #3170: Pretty print objects

cup, 2019-09-26:
Projects like Deno and Hermes can print Set objects in a useful representation:

~~~
$ deno
> new Set([10, 20])
Set { 10, 20 }
~~~

~~~
$ hermes-repl
>> new Set([10, 20])
Set { 10, 20 }
~~~

However Jerry output is of limited usefulness:

~~~
$ jerry
jerry> new Set([10, 20])
[object Set]
~~~

rerobika, 2019-09-30:
@cup Please note that the default `print` handler is just a basic implementation to print JavaScript values and you can modify/reimplement it if needed. However, you are right about that the current implementation does not distinguish the special objects (Array, Map, Set, etc.) So I mark this issue as a feature request for now.

AbhinavMir, 2020-03-22:
Hi, is this issue handled? Or should I quickly impement it?

rerobika, 2020-03-22:
Hi @AbhinavMir.

This issue has not been handled yet, so feel free to contribute, it's a good first issue :)

AbhinavMir, 2020-03-24:
I'd love some help getting around the initial codebase so as to keep my contribution consistent with the repo, where can I reach you?

rerobika, 2020-03-24:
I think this issue is a good place to discuss any questions. You should extend the functionalities of https://github.com/jerryscript-project/jerryscript/blob/master/jerry-ext/handler/handler-print.c#L40 this method, using the jerry API (https://jerryscript.net/api-reference/). Please keep in mind that the engine supports both the ES5.1 and the ES2015 standard, so I highly recommend the usage of https://jerryscript.net/api-reference/#jerry_is_feature_enabled.

AbhinavMir, 2020-03-26:
Hi, sorry for being late!
So I studied the structure of JerryScript for a bit, and had a few questions.

1. In Handler-Print.c's jerryx_handler_print, I need to work with the "return ret_value" right?
2. How can I test my code? When i go thru the API examples, I see it is mostly build examples. How can I test the actual JerryScript on Ubuntu 18.04?

galpeter, 2020-03-27:
Hi @AbhinavMir ,

1) The `ret_value` is not the one you want. It is used to return `undefined` or an error if something "bad" occurred.  You need `args_p` array which will contain the arguments passed for the `print` call.
Eg.: in case of the `print({data: 3 })`  js  the `args_p[0]` would be the first argument (the `{data:3}` object)

2) For testing there are multiple ways:
  * Use an example which registers the "print" handler: https://jerryscript.net/api-example/#example-6-using-jerryscript-extensions
  * Build the engine with the `./tools/build.py` build script and the `./build/bin/jerry` will be an example repl which uses the "print" handler.

3) In short yes. There are methods to query the type of an value and a container type of an object (if it is a container at all). One hard part is to iterate over the map/set elements as there is no direct api for it at the moment. A way to overcome this would be to get the JS "forEach" of the set/map prototype and invoke it with a custom callback method which prints out the elements.

AbhinavMir, 2020-03-27:
Thank you so much for your reply @galpeter !

closed 2021-12-06 by rerobika