jerryscript-project/jerryscript #3172: jerry_cleanup() do not call free callback assigned by jerry_set_object_native_pointer()

niklauslee, 2019-09-28:
Hello,

I want to make sure that `jerry_cleanup()` do not call free callback function assigned by `jerry_set_object_native_pointer()`.

My test procedure is:
1. Create a js object and then assign a native buffer to the object using `jerry_set_object_native_pointer()`.
2. Call `jerry_cleanup()`
3. The native buffer is not freed (free callback is not called).

I'm developing REPL and it supports `.reset` command to cleanup javascript context. It calls internally `jerry_cleanup()`. But the native buffers are not freed using `jerry_cleanup()`.

Is this intended for `jerry_cleanup()`?

If so, should I manage the native buffers internally and free them before calling `jerry_cleanup()`?.

Thanks in advance.

rerobika, 2019-09-30:
Hi @niklauslee!

I've investigated your problem and it's independent from the usage of the native pointers.
The problem is when you create an object with `jerry_create_object` it is not going to be freed until you do not call a `jerry_release_value` for the created object. Here comes a minimized example for your issue:

```c
#include "jerryscript.h"

int
main (void)
{
  jerry_init (JERRY_INIT_EMPTY);
  jerry_value_t object = jerry_create_object ();
  // do stuff
  
  // jerry_release_value should be called for object when it's no longer needed.
  jerry_cleanup ();
  return 0;
} /* main */
```
Return code: 120 (internal assertion, since the constructed object is not freed)

Extend the example with native pointers and free the object when it's no longer needed.

```c
#include "jerryscript.h"

static void
buffer_native_freecb (void *native_p)
{
  printf("Free is called\n");
  free (native_p);
}

static const jerry_object_native_info_t buffer_obj_type_info =
{
  .free_cb = buffer_native_freecb
};

int
main (void)
{
  jerry_init (JERRY_INIT_EMPTY);

  jerry_value_t object = jerry_create_object ();
  char *buffer_p = malloc (10 * sizeof (char));
  jerry_set_object_native_pointer (object, buffer_p, &buffer_obj_type_info);

  // do stuff

  jerry_release_value (object);
  jerry_cleanup ();
  return 0;
} /* main */
```
Output: `Free is called`
Return code: `0`

I hope it helps to resolve your problem! :)

niklauslee, 2019-10-01:
Hi @rerobika! Thank you for your answer with kind examples :)

Actually my problem in more complicated.

I have builtin class `GraphicContext` having internally a native graphic handle struct allocated by `malloc` and freed by callback function registered by `jerry_set_object_native_pointer`.

I also have a quite long js source code using the `GraphicContext` in a string and the code is executed by `jerry_run()`. After execution, instances of `GraphicContext` are remained in heap memory because they are referenced in the global object. So I can't call `jerry_release_value()` explicitly. And then, I wanted to clean up the JavaScript runtime to execute another source code. According to your answer, `jerry_cleanup()` do not call the free callbacks.

To sum up, is there a way to clean up all execution context with calling free callbacks?

If there isn't, I need to manage all the native graphic handles internally and free the all handles before calling `jerry_cleanup()`.

Thanks in advance.

galpeter, 2019-10-01:
Hi @niklauslee!

If an object is only referenced via the "global" object then `jerry_cleanup` will release it.

A bit extended example: The "demo" object is referenced only by the global object when the script is executed.
```c
#include <stdlib.h>

#include "jerryscript.h"

static void
buffer_native_freecb (void *native_p)
{
  printf("Free is called\n");
  free (native_p);
}

static const jerry_object_native_info_t buffer_obj_type_info =
{
  .free_cb = buffer_native_freecb
};

int
main (void)
{
  jerry_init (JERRY_INIT_EMPTY);

  jerry_value_t object = jerry_create_object ();
  char *buffer_p = malloc (10 * sizeof (char));
  jerry_set_object_native_pointer (object, buffer_p, &buffer_obj_type_info);

  jerry_value_t global = jerry_get_global_object ();
  jerry_value_t name = jerry_create_string ((const jerry_char_t *) "demo");
  jerry_value_t set_result = jerry_set_property (global, name, object);
  jerry_release_value (set_result);
  jerry_release_value (name);
  jerry_release_value (global);

  jerry_release_value (object); // If this is missing the free callback will not be invoked.

  // At this point the only reference to the "demo" object is from the global object
  const jerry_char_t script[] = "demo.value = 33; demo.value";
  jerry_value_t parsed_code = jerry_parse (NULL, 0, script, sizeof (script) - 1, JERRY_PARSE_NO_OPTS);

  if (!jerry_value_is_error (parsed_code))
  { 
    printf ("Running script\n");
    jerry_value_t run_result = jerry_run (parsed_code);
    if (jerry_value_is_error (run_result))
    { 
      printf ("Error during execution\n");
    }
    else
    {
      printf ("Result: %lf\n", jerry_get_number_value (run_result));
    }
    jerry_release_value (run_result);
  }

  jerry_release_value (parsed_code);

  jerry_cleanup ();
  return 0;
}
```

It is important to make sure that there is no reference from the C API to an object.

niklauslee, 2019-10-02:
Hi @galpeter!

Thank you for your answer. I have confirmed that `jerry_cleanup()` will call the free callbacks. I need to check there is no reference from C API.

Thanks again.

niklauslee, 2019-10-02:
I resolved this problem. Thank you.

closed 2019-10-02 by niklauslee