jerryscript-project/jerryscript #3214: Assertion context.status_flags & PARSER_SCANNING_SUCCESSFUL in parser_parse_source

renatahodovan, 2019-10-11:
###### JerryScript revision
84a56efd

###### Build platform
Linux-4.15.0-62-generic-x86_64-with-Ubuntu-18.04-bionic

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
var $ = { $ ( ){ } } = { g ( ) { ( ) => m ++ ( ) } }  = class extends C { constructor ( ) { super () } } 
```

###### Output
```text
ICE: Assertion 'context.status_flags & PARSER_SCANNING_SUCCESSFUL' failed at jerryscript/jerry-core/parser/js/js-parser.c(parser_parse_source):2555.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

###### Backtrace
```text
bt
#0  0xf7fd5079 in __kernel_vsyscall ()
#1  0xf77fc832 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf77fdcc1 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0x56579778 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-port/default/default-fatal.c:30
#4  0x5661440e in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x5661444f in jerry_assert_fail (assertion=0x566ae1c0 "context.status_flags & PARSER_SCANNING_SUCCESSFUL", file=0x566ade80 "jerryscript/jerry-core/parser/js/js-parser.c", function=0x56690340 <__func__.5511.lto_priv.462> "parser_parse_source", line=2555) at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x565d082b in parser_parse_source (arg_list_p=0x0, arg_list_size=0, source_p=0x566f3c00 <buffer.lto_priv> "var id_0 ; \nvar called = false ; \nfunction f ( ) { } \nvar B = class extends f { constructor ( ) { eval ( ) ; \nsuper ( id_1 ) \nthis .g ( ) \n} } \nC = class extends B { g ( ) { ( m ) => m ++ ( ) \n} } \nD "..., source_size=319, parse_opts=0, error_location_p=0xffffc6c0) at jerryscript/jerry-core/parser/js/js-parser.c:2555
#7  0x565d307d in parser_parse_script (arg_list_p=0x0, arg_list_size=0, source_p=0x566f3c00 <buffer.lto_priv> "var id_0 ; \nvar called = false ; \nfunction f ( ) { } \nvar B = class extends f { constructor ( ) { eval ( ) ; \nsuper ( id_1 ) \nthis .g ( ) \n} } \nC = class extends B { g ( ) { ( m ) => m ++ ( ) \n} } \nD "..., source_size=319, parse_opts=0, bytecode_data_p=0xffffc780) at jerryscript/jerry-core/parser/js/js-parser.c:3003
#8  0x5666bff8 in jerry_parse (resource_name_p=0xffffcd8e "/home/reni/.fuzzinator_12160//jerryscript/picireny/33680886534313296745214480367664733813.js", resource_name_length=92, source_p=0x566f3c00 <buffer.lto_priv> "var id_0 ; \nvar called = false ; \nfunction f ( ) { } \nvar B = class extends f { constructor ( ) { eval ( ) ; \nsuper ( id_1 ) \nthis .g ( ) \n} } \nC = class extends B { g ( ) { ( m ) => m ++ ( ) \n} } \nD "..., source_size=319, parse_opts=0) at jerryscript/jerry-core/api/jerry.c:426
#9  0x56668df1 in main (argc=2, argv=0xffffcb14) at jerryscript/jerry-main/main-unix.c:726
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

closed 2019-10-14 by zherczeg

ty5491003, 2019-10-14:
@renatahodovan Hello, i'm a fresh man of using jerryscript,  I noted that the "Backtrace" info in your issue, it's so useful! But when i compiled jerry on my pc, I can't find this. So i want to know how can i get the Backtrace info?  Thx!

renatahodovan, 2019-10-14:
Hi @ty5491003, you have to run JerryScript with gdb to get the backtrace:
```
> gdb --args ./build/bin/jerry test.js
...
(gdb) run
Starting program: jerryscript/build/bin/jerry test.js
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
ICE: Assertion 'context.status_flags & PARSER_SCANNING_SUCCESSFUL' failed at jerryscript/jerry-core/parser/js/js-parser.c(parser_parse_source):2555.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0xf7fd3939 in __kernel_vsyscall ()
(gdb) bt
#0  0xf7fd3939 in __kernel_vsyscall ()
#1  0xf77bf182 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf77a92b6 in abort () from /lib/i386-linux-gnu/libc.so.6
....
```

ty5491003, 2019-10-14:
@renatahodovan I'll try this, thx! 😆

YaoHouyou, 2019-10-14:
@renatahodovan 
I build jerryscript with the command above. 
````
python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
````
and run jerryscript with gdb to get backtrace, but there are no stack information. Do you known what went wrong?
````
> gdb --args ./build/bin/jerry /media/nisl/Elements/yhy/javascriptFuzzingOther/test/test.js
...
(gdb) run
Starting program: jerryscript/build/bin/jerry test.js
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
ICE: Assertion 'context.status_flags & PARSER_SCANNING_SUCCESSFUL' failed at jerryscript/jerry-core/parser/js/js-parser.c(parser_parse_source):2552.
Error: ERR_FAILED_INTERNAL_ASSERTION
[Inferior 1 (process 9062) exited with code 0170]
(gdb) bt
No stack.
````

YaoHouyou, 2019-10-15:
@renatahodovan 
I build jerryscript with the command above. 
````
python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
````
and run jerryscript with gdb to get backtrace, but there are no stack information. Do you known what went wrong? Thanks!
````
> gdb --args ./build/bin/jerry /media/nisl/Elements/yhy/javascriptFuzzingOther/test/test.js
...
(gdb) run
Starting program: jerryscript/build/bin/jerry test.js
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
ICE: Assertion 'context.status_flags & PARSER_SCANNING_SUCCESSFUL' failed at jerryscript/jerry-core/parser/js/js-parser.c(parser_parse_source):2552.
Error: ERR_FAILED_INTERNAL_ASSERTION
[Inferior 1 (process 9062) exited with code 0170]
(gdb) bt
No stack.
````

PR #3215 (2019-10-14): Set post-processing mode after an arrow expression scanning is completed
Fixes #3214.
commit f0e432c9 (2019-10-14): Set post-processing mode after an arrow expression scanning is completed (#3215)

Fixes #3214.

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com