jerryscript-project/jerryscript #3218: Garbage collection for native setInterval() implementation

niklauslee, 2019-10-15:
Hello,

I have my-own implementation of `setInterval(callback, interval)` written in C. When I run the below code, heap memory usage is increasing every callback call and finally system is stuck due to the lac of memory.

```js
setInterval(function () {
  var t = millis();
  console.log ("Current time is ", t, "ms.");
}, 100);
```

In our system, there is the main even loop. In the event loop, check interval time and call the callback using `jerry_call_function()`. Every callback call results memory consumption, but garbages are not collected even if `jerry_gc()` is called.

Like below code, heap memory is not increased if the callback is not generates garbages (like without `var t = millis();`).

```js
setInterval(function () {
  console.log ("Current time is unknown");
}, 100);
```

Same phenomenon is observed in all cases that javascript callback function is called using `jerry_call_function()` from C code side. (Timer callback, UART data read callback, etc.)

For more info:
- [setInterval() implementation](https://github.com/kameleon-project/kameleon/blob/master/src/global.c#L317)
- [the code calling callback](https://github.com/kameleon-project/kameleon/blob/master/src/global.c#L282)

What I'm doing wrong? Any advices?

Thank you.

rerobika, 2019-10-15:
Hi @niklauslee.

After a bit of investigation I've realized that there is no problem with the GC, the leak in your code comes from [this](https://github.com/kameleon-project/kameleon/blob/7fd18349f952fb6c91582e1a85a1eadb8e8243d9/src/jerryxx.c#L91) line, since the `toString`-ed value must be released.

Please let me know whether this missing `jerry_release_value` solved your problem or not. :)

Best regards,
Robert

niklauslee, 2019-10-15:
Hi @rerobika. Now it's works well after fixing what you pointed.
Thank you so much! 😄

closed 2019-10-15 by niklauslee