jerryscript-project/jerryscript #3229: An issue about method Number.prototype.toString

YaoHouyou, 2019-10-18:
According to ES5.1, for method [Number.prototype.toString ( [ radix ] )](http://www.ecma-international.org/ecma-262/5.1/index.html#sec-15.7.4.2), if ToInteger(radix) is not an integer between 2 and 36 inclusive throw a RangeError exception. But, Jerry did not throw a RangeError exception when running the following test case.

###### JerryScript revision
[7df87b7](https://github.com/jerryscript-project/jerryscript/commit/7df87b7778124af9b0dd309570c33864a67792f7)

###### Build platform
```
Ubuntu 16.04.6 LTS (Linux 4.15.0-55-generic x86_64)
```

###### Build steps
```
python tools/build.py --clean --debug --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --mem-heap=1572864 --profile=es2015-subset
```

###### Test case
```
var NISLFuzzingFunc = function(e) {
    var tmp = Number(0).toString(e);
    print(tmp);
};
var NISLParameter4 = "hello jerry";
NISLFuzzingFunc(NISLParameter4);
```

###### Execution steps
```
build/bin/jerry testcase.js
```

###### Output
`0`

###### Expected behavior
Throw a RangeError exception

rerobika, 2019-10-18:
#3224  Should resolve this issue.

Yuyupo, 2019-10-18:
This issue is unrelated to the mentioned Pull Request. If all goes well, the commit above should fix it.

PR #3232 (2019-10-29): Fix issue about Number.prototype.toString
Fixes #3229

JerryScript-DCO-1.0-Signed-off-by: Daniella Barsony bella@inf.u-szeged.hu
commit 11818f1c (2019-10-29): Fix issue about Number.prototype.toString (#3232)

Fixes #3229

JerryScript-DCO-1.0-Signed-off-by: Daniella Barsony bella@inf.u-szeged.hu
closed 2019-10-29 by rerobika