jerryscript-project/jerryscript #3247: Assertion ecma_is_value_object (value) in ecma_get_object_from_value

renatahodovan, 2019-10-22:
###### JerryScript revision
59e0d6e2

###### Build platform
Linux-4.15.0-62-generic-x86_64-with-Ubuntu-18.04-bionic

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
(new Int8Array((new ArrayBuffer), 1, Infinity)).map(function() { })
```

###### Output
```text
ICE: Assertion 'ecma_is_value_object (value)' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_get_object_from_value):749.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

###### Backtrace
```text
bt
#0  0xf7fd5079 in __kernel_vsyscall ()
#1  0xf77fc832 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf77fdcc1 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0x56579ec6 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-port/default/default-fatal.c:30
#4  0x56613040 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x56613081 in jerry_assert_fail (assertion=0x566c7e40 "ecma_is_value_object (value)", file=0x566c7820 "jerryscript/jerry-core/ecma/base/ecma-helpers-value.c", function=0x5668ae20 <__func__.4247.lto_priv.711> "ecma_get_object_from_value", line=749) at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x56658cd2 in ecma_get_object_from_value (value=24) at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:749
#7  0x56611de3 in ecma_op_create_typedarray_with_type_and_length (obj_p=0xf5f02080, array_length=4294967295) at jerryscript/jerry-core/ecma/operations/ecma-typedarray-object.c:1194
#8  0x5662b54d in ecma_builtin_typedarray_prototype_map (this_arg=4126154883, cb_func_val=4126154739, cb_this_arg=72) at jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.c:454
#9  0x5662a2ba in ecma_builtin_typedarray_prototype_dispatch_routine (builtin_routine_id=82, this_arg_value=4126154883, arguments_list=0xffffc400, arguments_number=1) at jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.inc.h:62
#10 0x56629524 in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_TYPEDARRAY_PROTOTYPE, builtin_routine_id=82, this_arg_value=4126154883, arguments_list_p=0xffffc400, arguments_list_len=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1021
#11 0x56629785 in ecma_builtin_dispatch_call (obj_p=0xf5f02050, this_arg_value=4126154883, arguments_list_p=0xffffc698, arguments_list_len=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1046
#12 0x566401ec in ecma_op_function_call (func_obj_p=0xf5f02050, this_arg_value=4126154883, arguments_list_p=0xffffc698, arguments_list_len=1) at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:729
#13 0x565e5200 in opfunc_call (frame_ctx_p=0xffffc710) at jerryscript/jerry-core/vm/vm.c:581
#14 0x565f63fc in vm_execute (frame_ctx_p=0xffffc710, arg_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:3604
#15 0x565f6cff in vm_run (bytecode_header_p=0xf4f03680, this_binding_value=4126149459, lex_env_p=0xf5d007b0, parse_opts=0, arg_list_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:3724
#16 0x565e44a9 in vm_run_global (bytecode_p=0xf4f03680) at jerryscript/jerry-core/vm/vm.c:282
#17 0x56671f58 in jerry_run (func_val=4126147539) at jerryscript/jerry-core/api/jerry.c:576
#18 0x5666e6a0 in main (argc=2, argv=0xffffcb44) at jerryscript/jerry-main/main-unix.c:739
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [JsProFuzz](https://www.researchgate.net/publication/303482425_Fuzzing_JavaScript_Engine_APIs). </sup>

yorkie, 2019-10-23:
Could not reproduce the problem, it prints the following `RangeError` on my machine:

```shell
$ ./build/bin/jerry ./test.js
Script Error: RangeError: Invalid length.
```

akosthekiss, 2019-10-23:
@yorkie "on my machine" is somewhat vague. What is your platform? Also: did you use the build steps that are given in the issue report, and what is your JerryScript version (hash)? Please, provide as much information as the original report.

yorkie, 2019-10-23:
Sorry about that @akosthekiss, the following is what my environment exactly is:

```bash
$ uname -av
Darwin yorkie-on-rokid.local 17.2.0 Darwin Kernel Version 17.2.0: Fri Sep 29 18:27:05 PDT 2017; root:xnu-4570.20.62~3/RELEASE_X86_64 x86_64

$ ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --strip=off --system-allocator=on --logging=on  --error-messages=on --profile=es2015-subset
$ ./build/bin/jerry # with the given test case
Script Error: RangeError: Invalid length.

$ git log -1
commit 59e0d6e262980df321448e78b37982f1807031bc (HEAD -> master, origin/master, origin/HEAD)
Author: Csaba Osztrogonác <oszi@inf.u-szeged.hu>
Date:   Tue Oct 22 18:46:56 2019 +0200

    Make snapshot unittest easily updatable (#3242)

    JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác oszi@inf.u-szeged.hu
```

yorkie, 2019-10-23:
Oh, I did apply a hidden patch(#3244), I confirmed it also fixed this, so it's duplicated with #3243.

rerobika, 2019-10-25:
Duplicate of #3243, closed via #3244.

closed 2019-10-25 by rerobika