jerryscript-project/jerryscript #3254: Reallocating memory of context

rafail-ufa, 2019-10-27:
Hello!
At this moment I've tried to make dynamic memory allocation. Scripts that will be used are static, but i don't know exact memory consumption of them. Now memory allocation works like in tutorials:

```
void * context_alloc(size_t size, void *cb_data_p)
{
	return malloc(size);
}

jerry_context_t* create_context(size_t context_size_in_KB)
{
	return jerry_create_context(context_size_in_KB * 1024, context_alloc, NULL);;
}  

void initial_context_create(size_t context_size_in_KB)
{
	_context_pointer = create_context(context_size_in_KB);
	jerry_port_default_set_current_context(_context_pointer);
}

jerry_context_t* _context_pointer;

main()
{
	initial_context_create(1);
	jerry_init(JERRY_INIT_MEM_STATS);
}
```
And tried to do something like this:

```
void * context_alloc(size_t size, void *cb_data_p)
{
	return malloc(size);
}

jerry_context_t* create_context(size_t context_size_in_KB)
{
	return jerry_create_context(context_size_in_KB * 1024, context_alloc, NULL);;
}  

void initial_context_create(size_t context_size_in_KB)
{
	_context_pointer = create_context(context_size_in_KB);
	jerry_port_default_set_current_context(_context_pointer);
}

void * jsrt::jsrt_runner::resized_context_alloc(size_t size, void * prev_ptr)
{
	return realloc(prev_ptr, size);
}

jerry_context_t* create_resized_context(size_t new_context_size_in_KB)
{
	return jerry_create_context(new_context_size_in_KB * 1024, resized_context_alloc, _context_pointer);;
}

void enlarge_context(size_t new_context_size)
{
	_context_pointer = create_resized_context(new_context_size);
}

jerry_context_t* _context_pointer;

main()
{
	initial_context_create(1);
	jerry_init(JERRY_INIT_MEM_STATS);
	
	enlarge_context(2);
	
	/* after that i should do something to reinitialize or edit enlarged _context_pointer,
	because it doesn't in it's internal structure heap_size is 1024 (as set previously)
	and i can not use enlarged allocated momery for the jerry_context_t (_context_pointer)
	*/
	
	/*
	if i try to do jerry_init() again, of course all allocated memory is available, but i'm loosing all context data (such as ecma objects, and other previously created objects and variables before enlarging). Because before jerry_init() i should do jerry_cleanup()
	*/
}
```

Is there any possibility to enlarge existing jerry_context_t without loosing all the data.
Thank You very much!

rerobika, 2019-10-28:
Hi @rafail-ufa !

Your problem points to the fact that currently the internal heap pointer management is not prepared for the reallocation of the heap. So, unfortunately there is no way to enlarge the existing context  without loosing all the data at the moment. This change would require a huge amount of refactoring and I'm not sure that we have capacity to do it now days. However, my suggestion is to use the system-allocator option. `python tools/build.py --system-allocator ON` (`--compile-flag='-m32'` build option is potentially needed on 64bit architecture.) Via these options there is not need to take care of reallocating the heap/context.

rafail-ufa, 2019-10-28:
Thank You @rerobika  for quick answer!
This is really the solution for that situation, I've tested in single instance. Will it work for multithread situation? I mean that have several threads with independent contexts for each, and should I change context allocation method, or it's OK and will work properly.

rerobika, 2019-10-28:
@rafail-ufa Yes it should definitely work. To give some hints [here](https://gist.github.com/rtakacs/3948e3294bb05fd9c74f1566f831c2ec) is an example for the multi-threaded context initialization.

> should I change context allocation method, or it's OK and will work properly.

The context allocation method can be set for each context instance independently with a [callback](https://gist.github.com/rerobika/79a59c06e4b5635a5247d1e47ebab2a9#file-pthread-c-L24) function, but almost always the single context allocator callback can be used in case of multiple-context (until the allocator function ensures that the instances are not overlapping, e.g. using malloc is always a good point to start).

rafail-ufa, 2019-11-04:
@rerobika Thank you, it really helped!
Are there any plans to implement system-allocator option for 64 bit build in future?

rerobika, 2019-11-04:
@rafail-ufa Yes, it's already on our TODO list, but the time of the implementation is unclear yet.

rafail-ufa, 2019-11-05:
@rerobika That's good news, thank a lot for information and help! It really makes sense for me.

closed 2019-11-07 by rafail-ufa