jerryscript-project/jerryscript #3255: error when run api-example-1 undefined reference to `jerry_run_simple'

MaJun0007, 2019-10-28:
I tried to run example on ubuntu 16.04. I got a api-example-1 .c

#include "jerryscript.h"

int
main (void)
{
  const jerry_char_t script[] = "var str = 'Hello, World!';";

  bool ret_value = jerry_run_simple (script, sizeof (script) - 1, JERRY_INIT_EMPTY);

  return (ret_value ? 0 : 1);
}

then run
$ gcc api-example-1.c -o api-example-1 $(pkg-config --cflags --libs libjerry-core libjerry-port-default libjerry-libm)
api-example-1.c:(.text+0x5d): undefined reference to `jerry_run_simple'
collect2: error: ld returned 1 exit status

rerobika, 2019-10-28:
Hi @MaJun0007,

Could you please check whether you followed correctly the [build-steps](https://github.com/jerryscript-project/jerryscript/blob/master/docs/03.API-EXAMPLE.md#before-trying-out-the-examples-get-and-build-jerryscript-library)? I've just copy-pasted the commands from the documentation and it worked for me. Here is my [log](https://gist.github.com/rerobika/7c2a7dd03d587e436a52332649ee5f06) from the configuration/build. Please compare this result with yours, and feel free to share your output if some relevant part of the logs are differ.

MaJun0007, 2019-10-29:
@rerobika thank you 

I followed commands:
```
$ mkdir jerry
$ cd jerry
$ git clone https://github.com/jerryscript-project/jerryscript.git
$ jerryscript/tools/build.py --builddir=$(pwd)/example_build --cmake-param="-DCMAKE_INSTALL_PREFIX=$(pwd)/example_install/"
$ make -C $(pwd)/example_build install
```

but still get error :
`api-example-1.c:(.text+0x5d): undefined reference to 'jerry_run_simple'`

here is my log :
```
root@Majun:/var/jerry# git clone https://github.com/jerryscript-project/jerryscript.git
Cloning into 'jerryscript'...
remote: Enumerating objects: 49, done.
remote: Counting objects: 100% (49/49), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 43053 (delta 22), reused 11 (delta 4), pack-reused 43004
Receiving objects: 100% (43053/43053), 47.61 MiB | 11.03 MiB/s, done.
Resolving deltas: 100% (33646/33646), done.
Checking connectivity... done.
root@Majun:/var/jerry# jerryscript/tools/build.py --builddir=$(pwd)/example_build --cmake-param="-DCMAKE_INSTALL_PREFIX=$(pwd)/example_install/"
-- The C compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- CMAKE_BUILD_TYPE            MinSizeRel
-- CMAKE_C_COMPILER_ID         GNU
-- CMAKE_SYSTEM_NAME           Linux
-- CMAKE_SYSTEM_PROCESSOR      x86_64
-- BUILD_SHARED_LIBS           OFF
-- ENABLE_LTO                  ON
-- ENABLE_STRIP                ON
-- JERRY_CMDLINE               ON
-- JERRY_CMDLINE_TEST          OFF
-- JERRY_CMDLINE_SNAPSHOT      OFF
-- JERRY_LIBFUZZER             OFF (FORCED BY COMPILER)
-- JERRY_PORT_DEFAULT          ON (FORCED BY CMDLINE OR LIBFUZZER OR TESTS)
-- JERRY_EXT                   ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_LIBM                  ON
-- UNITTESTS                   OFF
-- DOCTESTS                    OFF
-- ENABLE_ALL_IN_ONE              OFF
-- JERRY_CPOINTER_32_BIT          OFF
-- JERRY_DEBUGGER                 OFF
-- JERRY_ERROR_MESSAGES           OFF
-- JERRY_EXTERNAL_CONTEXT         OFF
-- JERRY_PARSER                   ON
-- JERRY_LINE_INFO                OFF
-- JERRY_LOGGING                  OFF
-- JERRY_MEM_STATS                OFF
-- JERRY_MEM_GC_BEFORE_EACH_ALLOC OFF
-- JERRY_PARSER_DUMP_BYTE_CODE    OFF
-- JERRY_PROFILE                  es5.1
-- JERRY_REGEXP_STRICT_MODE       OFF
-- JERRY_REGEXP_DUMP_BYTE_CODE    OFF
-- JERRY_SNAPSHOT_EXEC            OFF
-- JERRY_SNAPSHOT_SAVE            OFF
-- JERRY_SYSTEM_ALLOCATOR         OFF
-- JERRY_VALGRIND                 OFF
-- JERRY_VM_EXEC_STOP             OFF
-- JERRY_GLOBAL_HEAP_SIZE         (512)
-- JERRY_GC_LIMIT                 (0)
[... 279 lines cut ...]
-- Installing: /var/jerry/example_install/include/jerryscript-ext/module.h
-- Installing: /var/jerry/example_install/include/jerryscript-ext/arg.h
-- Installing: /var/jerry/example_install/include/jerryscript-ext/autorelease.impl.h
-- Installing: /var/jerry/example_install/include/jerryscript-ext/autorelease.h
-- Installing: /var/jerry/example_install/include/jerryscript-ext/handle-scope.h
-- Installing: /var/jerry/example_install/include/jerryscript-ext/debugger.h
-- Installing: /var/jerry/example_install/include/jerryscript-ext/handler.h
-- Installing: /var/jerry/example_install/include/jerryscript-ext/arg.impl.h
-- Installing: /var/jerry/example_install/lib/libjerry-port-default.a
-- Installing: /var/jerry/example_install/lib/libjerry-port-default-minimal.a
-- Installing: /var/jerry/example_install/lib/pkgconfig/libjerry-port-default.pc
-- Installing: /var/jerry/example_install/lib/pkgconfig/libjerry-port-default-minimal.pc
-- Up-to-date: /var/jerry/example_install/include
-- Installing: /var/jerry/example_install/include/jerryscript-port-default.h
-- Installing: /var/jerry/example_install/bin/jerry
make: Leaving directory `/var/jerry/example_build'
root@Majun:/var/jerry# export PKG_CONFIG_PATH=$(pwd)/example_install/lib/pkgconfig/
root@Majun:/var/jerry# pkg-config --cflags --libs libjerry-core libjerry-port-default libjerry-ext libjerry-libm
-I/var/jerry/example_install/include -I/var/jerry/example_install/include/jerry-libm  -L/var/jerry/example_install/lib -ljerry-core -ljerry-port-default -ljerry-ext -ljerry-libm
root@Majun:/var/jerry# vim api-example-1.c
root@Majun:/var/jerry# gcc api-example-1.c -o api-example-1 $(pkg-config --cflags --libs libjerry-core libjerry-port-default libjerry-libm)
/tmp/ccNV4rj9.o: In function `main':
api-example-1.c:(.text+0x5d): undefined reference to `jerry_run_simple'
collect2: error: ld returned 1 exit status
```

rerobika, 2019-10-29:
cc @galpeter .

galpeter, 2019-10-30:
Hi @MaJun0007 

It seems the problem is the default `lto` configuration in the project. On older gcc (pre 5.0) the linker will not find the symbols correctly if a library is built with lto.

What you can do at the moment is two things:
1) Disable the lto during the JerryScript build. To to this add the `--lto=off` option to the `build.py` command.
2) Add the `-flto` option when building the examples.

We'll update the documentation to highlight this. Thanks for reporting!

rerobika, 2019-11-06:
Resolved via #3274.

closed 2019-11-06 by rerobika