jerryscript-project/jerryscript #3277: A possible performance issue about 'for' loop

YuanWangC, 2019-11-02:
When running the testcases below, jerryscript is hundred times slower than other engines with the same output. Maybe, it is an issue of jerryscript.
## Version  
[9ab4872](https://github.com/jerryscript-project/jerryscript/commit/9ab48722445c483a2b984fef609d13ab87894dea)
## Build step
```
python tools/build.py --clean --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --mem-heap=1572864 --profile=es2015-subset
```
## Testcase
```
var NISLFuzzingFunc = function(t) {
    var counter = 0;
    for (var i = 0; i < t ; i += 1) {
        counter += 1;
    }
    print (counter);
};
var NISLParameter = 594755575;
NISLFuzzingFunc(NISLParameter);
```
## Command  
`jerryscript/build/bin/jerry testcase.js`
## Excution time  
jerryscript: `55602ms`
V8: `486ms`
spiderMonkey: `601ms`
chakra: `343ms`
javascriptCore: `477ms`
## Output
`594755575`

rerobika, 2019-11-04:
Hi @YuanWangC,

There isn't any performance issue in the related test-case and the comparison seems a bit pointless for me since the listed other engines' design/attributes/field of application/memory consumption/binary size/execution speed completely differs from JerryScript's. JerryScript is designed for low-end devices with minimal memory footprint/binary size with a byte-code interpreter virtual machine. The other (browser) engines are highly optimized for execution speed and they are using JIT (Just In Time) compilation for JS execution.

rerobika, 2019-11-06:
I see no objection to the reasoning, I'm closing the issue, please reopen it if needed.

closed 2019-11-06 by rerobika