jerryscript-project/jerryscript #3288: Fixing "sudo" usage for dependency installation

ScooloV, 2019-11-07:
In [tools] directory there are several *.sh scripts that are supposed to work for debian/ubuntu dependency installation for lib building.

Let's take **apt-get-install-deps.sh**
When I try to build in jenkins/docker I get this error (I use **FROM debian:stretch**):
```
[2019-11-07T04:44:32.865Z] [91mtools/apt-get-install-deps.sh: line 17: sudo: command not found
[2019-11-07T04:44:32.866Z] [0m[91mtools/apt-get-install-deps.sh: line 18: sudo: command not found
```
Reason is simple, since it is docker, you are root by the default, and root requires no sudo (and sudo is not installed by the default).

Yes we can install sudo earlier and/or add the user to the sudo group, but there is also solution I use for another scripts, which are building another libraries:

```
#!/bin/bash

if [ "$(whoami)" != "root" ]; then
    SUDO=sudo
fi

# Copyright JS Foundation and other contributors, http://js.foundation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

${SUDO} apt-get update -q
${SUDO} apt-get install -q -y \
    make cmake \
    gcc gcc-multilib \
    doxygen \
    cppcheck vera++ python pylint python-serial
```

May be it would be a good idea to add that to all *.sh scripts?

rerobika, 2019-11-11:
Hi @ScooloV!

The suggestion seems reasonable/easy task to me. Would you like to submit a PR for it?! :)

ScooloV, 2019-11-11:
@rerobika I think there is no need for my personal PR for that, just change by yourself, if possible.

PR #3315 (2019-11-14): Fixing sudo usage for dependency installation
Fixes #3288, the fix is same as the bug reporter suggested.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác oszi@inf.u-szeged.hu
commit 84f60c19 (2019-11-14): Fixing sudo usage for dependency installation (#3315)

Fixes #3288, the fix is same as the bug reporter suggested.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác oszi@inf.u-szeged.hu
closed 2019-11-14 by zherczeg