jerryscript-project/jerryscript #3303: Assertion !(type & SCANNER_LITERAL_IS_VAR) in scanner_pop_literal_pool

renatahodovan, 2019-11-12:
###### JerryScript revision
452c7818

###### Build platform
Linux-4.15.0-62-generic-x86_64-with-Ubuntu-18.04-bionic

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
try { } catch ( e ) {
  if ($) { var e }
}
```

###### Output
```text
ICE: Assertion '!(type & SCANNER_LITERAL_IS_VAR)' failed at jerryscript/jerry-core/parser/js/js-scanner-util.c(scanner_pop_literal_pool):546.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

###### Backtrace
```text
bt
#0  0xf7fd5079 in __kernel_vsyscall ()
#1  0xf77fc832 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf77fdcc1 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0x5657a737 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-port/default/default-fatal.c:30
#4  0x56604cd3 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x56604d14 in jerry_assert_fail (assertion=0x566c9fa0 "!(type & SCANNER_LITERAL_IS_VAR)", file=0x566c9c40 "jerryscript/jerry-core/parser/js/js-scanner-util.c", function=0x566a3ae0 <__func__.5619.lto_priv.463> "scanner_pop_literal_pool", line=546) at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x56617b4b in scanner_pop_literal_pool (context_p=0xffffc4c0, scanner_context_p=0xffffc300) at jerryscript/jerry-core/parser/js/js-scanner-util.c:546
#7  0x56625489 in scanner_scan_statement_end (context_p=0xffffc4c0, scanner_context_p=0xffffc300, type=LEXER_RIGHT_BRACE) at jerryscript/jerry-core/parser/js/js-scanner.c:1815
#8  0x56627127 in scanner_scan_all (context_p=0xffffc4c0, arg_list_p=0x0, arg_list_end_p=0x0, source_p=0x5670bae0 <buffer.lto_priv> "function id_0 ( ) { } \nvar x = function ( ) { throw \"x\" ; \n} ; \nvar y = function ( ) { throw \"y\" ; \n} ; \ntry { id_1 ( x ( ) , y ( ) ) ; \nprint ( false ) ; \n} catch ( e ) { if ( e === \"y\" ) { var e ; \n"..., source_end_p=0x5670bbe4 <buffer.lto_priv+260> "") at jerryscript/jerry-core/parser/js/js-scanner.c:2273
#9  0x56611832 in parser_parse_source (arg_list_p=0x0, arg_list_size=0, source_p=0x5670bae0 <buffer.lto_priv> "function id_0 ( ) { } \nvar x = function ( ) { throw \"x\" ; \n} ; \nvar y = function ( ) { throw \"y\" ; \n} ; \ntry { id_1 ( x ( ) , y ( ) ) ; \nprint ( false ) ; \n} catch ( e ) { if ( e === \"y\" ) { var e ; \n"..., source_size=260, parse_opts=0, error_location_p=0xffffc6d0) at jerryscript/jerry-core/parser/js/js-parser.c:1840
#10 0x56615741 in parser_parse_script (arg_list_p=0x0, arg_list_size=0, source_p=0x5670bae0 <buffer.lto_priv> "function id_0 ( ) { } \nvar x = function ( ) { throw \"x\" ; \n} ; \nvar y = function ( ) { throw \"y\" ; \n} ; \ntry { id_1 ( x ( ) , y ( ) ) ; \nprint ( false ) ; \n} catch ( e ) { if ( e === \"y\" ) { var e ; \n"..., source_size=260, parse_opts=0, bytecode_data_p=0xffffc790) at jerryscript/jerry-core/parser/js/js-parser.c:2443
#11 0x56676151 in jerry_parse (resource_name_p=0xffffcd98 "/home/reni/.fuzzinator_14762//jerryscript/picireny/304041482045963303880035536015562381295.js", resource_name_length=93, source_p=0x5670bae0 <buffer.lto_priv> "function id_0 ( ) { } \nvar x = function ( ) { throw \"x\" ; \n} ; \nvar y = function ( ) { throw \"y\" ; \n} ; \ntry { id_1 ( x ( ) , y ( ) ) ; \nprint ( false ) ; \n} catch ( e ) { if ( e === \"y\" ) { var e ; \n"..., source_size=260, parse_opts=0) at jerryscript/jerry-core/api/jerry.c:445
#12 0x56672ed7 in main (argc=2, argv=0xffffcb24) at jerryscript/jerry-main/main-unix.c:731
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

zherczeg, 2019-11-27:
I don't see this error anymore. Could you check whether it is still valid?

renatahodovan, 2019-11-27:
@zherczeg After a short bisecting it seems that you fixed it in #3322 (at least the assertion failure disappeared from that commit).

closed 2019-11-27 by renatahodovan