jerryscript-project/jerryscript #3393: A question about getting arguments' descriptor of strict mode functions

YiWen-y, 2019-11-30:
Strict mode functions are not allowed to access property `arguments`. However, when running this testcase blow, the output is  `[object Object] `.

###### JerryScript revision
[996bf76](https://github.com/jerryscript-project/jerryscript/commit/996bf76f5956c65342c1726d3f629fbea41320ab)

###### Build platform
    Ubuntu 16.04.6 LTS (Linux 4.15.0-55-generic x86_64)

###### Build steps
    python tools/build.py --clean --debug --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --mem-heap=1572864 --profile=es2015-subset

###### Test case
```
var NISLFuzzingFunc = function() {
    print(Object.getOwnPropertyDescriptor(new Function("'use strict';"), "arguments"));
};
NISLFuzzingFunc();
```

###### Execution steps
    build/bin/jerry testcase.js

###### Output
    [object Object]

###### Expected behavior
    undefined

zherczeg, 2019-11-30:
```
(function () {})
()
arguments: null
caller: null
length: 0
name: ""
prototype: Object { … }
<prototype>: function ()

(function () { "use strict" })
()
length: 0
name: ""
prototype: Object { … }
<prototype>: function ()
```

Looks like they don't have caller as well.

YaoHouyou, 2019-12-02:
@zherczeg Can you explain why the output is `[object Object]` when getting the description of the `arguments`?  Thanks！
In addition, maybe, the output of other engines  can be used as a reference. Such as v8, spiderMonkey, chakra, javascriptCore, whose outputs are `undefined`.

rerobika, 2020-01-22:
Hi @YaoHouyou!

Sorry for the late answer. We haven't updated the `arguments` and `caller` properties behavior to according to the ES6 spec yet. However, I've just create a patch which contains these changes. Feel free to take a look at it and indicate whether it completely resolves your problem.

PR #3530 (2020-01-23): Update "arguments" and "caller" properties of function object to conform ES6 changes
This patch resolves #3393.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 3c5fb342 (2020-01-23): Update "arguments" and "caller" properties of function object to conform ES6 changes (#3530)

This patch resolves #3393.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2020-01-23 by rerobika