jerryscript-project/jerryscript #3432: Tick/stepping of execution for embedded systems with no OS

dazhbog, 2019-12-09:
Hi,

I am running Jerryscript on a microcontroller and I was wondering if there is a way to step the execution of JS from C. 

For example lets say the Javascript is busy going around in a loop like while(1), could the C program dictate the speed of execution? Like, instead of 
```
jerry_run(parsed_code);
```
we could have 
```
jerry_load(parsed_code);
while(1){
 jerry_next_tick();
 // Do other things here, like handle Bluetooth, I/O, sleep, etc.
}
```

Alternatively if this is not impossible, is there a way to either have a yield call from JS or a pause one from C? 

Similar issue #1356 

Thanks.

zherczeg, 2019-12-09:
You cannot suspend the execution of JS (except for generator functions), but there are periodic calls to a C function if you enable the `JERRY_VM_EXEC_STOP` feature:

https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/vm/vm.c#L1262

Just do what you need, and return with undefined after you are done.

dazhbog, 2019-12-09:
OK, thanks @zherczeg. Will give it a go.

closed 2019-12-09 by dazhbog