jerryscript-project/jerryscript #3473: A question of function parameter

YuanWangC, 2019-12-20:
When executing this testcase on other engines, (like [quickjs](https://bellard.org/quickjs/), v8 and spiderMonkey) I get the expected output. Only jerryscript throws an exception. Unfortunately I didn't find clear description about it in ES5.1 or ES6. Is this an issue of jerryscript?

###### JerryScript revision
[996bf76](https://github.com/jerryscript-project/jerryscript/commit/996bf76f5956c65342c1726d3f629fbea41320ab)
###### Test case
```
var NISLFuzzingFunc = function(name,){
     print(arguments.length);
};
NISLFuzzingFunc(1,2,3);
```
###### Output
`
Script Error: SyntaxError: Identifier expected.
`
###### Expected Output
`
3
`

zherczeg, 2019-12-20:
They changed FormalParameters definition between 7.0 and 8.0:

https://www.ecma-international.org/ecma-262/7.0/index.html#prod-FormalParameters
https://www.ecma-international.org/ecma-262/8.0/index.html#prod-FormalParameters

The new variant allows a comma at the end.

This is fairly easy to implement, are you interested?

akosthekiss, 2019-12-23:
The project is still heading towards ES6 compatibility. Is it relevant to focus on ES7 and/or 8 features at the moment?

rerobika, 2020-01-16:
Resolved via #3477.

closed 2020-01-16 by rerobika