jerryscript-project/jerryscript #3477: A question of Assignment

zhangle1987, 2019-12-23:
when execute this case on the other engines can be run success, but jerryscript throws exception:Script Error: ReferenceError
expression like this:

`var _ref;
var args = [1];
function Root() {

}

function _p(arg1, arg2) {
  return arg1 ? arg1 : arg2;
}

//success
var temp = (_ref = Object.getPrototypeOf(Root)).call({});
var constructor = _p(this,temp);

// fail on jerryscript
var constructor = _p(this,
  (_ref = Object.getPrototypeOf(Root)).call({})
);`

zherczeg, 2019-12-23:
I asked in #3473, but do you interested to work on these?

rerobika, 2020-01-16:
@zherczeg IMO #3473 has nothing in common with this issue. The problem is the wrong opcode combining during the argument list parsing.

PR #3525 (2020-01-17): Fix assignment opcode transformation for CBC_PUSH_THIS_LITERAL
This patch fixes #3477.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 71d4e121 (2020-01-17): Fix assignment opcode transformation for CBC_PUSH_THIS_LITERAL (#3525)

This patch fixes #3477.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2020-01-17 by dbatyai