jerryscript-project/jerryscript #3500: [question] jerry_release_value() for jerry_create_undefined()?

dxjia, 2020-01-08:
Hi , all
I have noted many jerry APIs are marked as `release_value` needed for the returned value of those API, for example, [jerry_create_array](https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/api/jerry.c#L1372).

And for some APIs, such as `jerry_create_undefined()`, that note is not declared in code, but in some documents, the sample code is showing the jerry_release_value() is needed for them.
For example, the 'jerry_create_bool()' mentioned [here](https://github.com/jerryscript-project/jerryscript/blob/master/docs/06.REFERENCE-COUNTING.md).

Just want to confirm, if we called jerry_create_undefined()/jerry_create_bool() by ourself, should we remember to call **jerry_release_value** to release it somewhere?

zherczeg, 2020-01-08:
The general policy is, that you need to release all created values. Currently you don't need to release certain values, because the engine internally does not allocate memory for them, but this can be changed in the future, and old code should still run.

dxjia, 2020-01-08:
@zherczeg

dxjia, 2020-01-08:
@zherczeg 
Thanks for so quick response.
So if I give a `jerry_value_t` variable with `jerry_create_undefined()` as its initialization value, it's better to call `jerry_release_value()` on it before assign another jerry_value_t to it, right?

zherczeg, 2020-01-08:
yes

dxjia, 2020-01-09:
@zherczeg 
Cleared enough. Thank you so much.

closed 2020-01-09 by dxjia