jerryscript-project/jerryscript #3526: Storing result of "super" call in local var breaks "this" context.

iborusik, 2020-01-17:
Jerry version:

Rev: 71d4e121
Build command: tools/build.py --profile=es2015-subset ---error-messages ON

OS:
Mac OS 10.14.2

**Code sample:**
```
class Base
{
    constructor()
    {
    this.x = [0, 0];
    }

    sampeFunc()
    {
        this.x[0]++; // crash here
        return this.x;
    }
}

class Extended extends Base
{
    sampeFunc()
    {
        var s = super.sampeFunc(); 
        return s;  // ok if change to "return super.sampleFunc()"
    }
}
var s = new Extended();
s.sampeFunc(); // :( oops

```

**ER:**
No crash shouldn't appear.

**AR**
Crash:  _TypeError: Cannot read property '0' of undefined_

Comments:
This issue can be related to [this issue](https://github.com/jerryscript-project/jerryscript/issues/2671)
As far as I understand line "var s = super.sampleFunc()" should emit "CBC_EXT_SUPER_PROP_CALL". But it doesn't pass this condition "!LEXER_IS_BINARY_LVALUE_TOKEN (context_p->stack_top_uint8)" : js-parser-expr.c:1470

rerobika, 2020-01-17:
Hi @iborusik!

Thanks for the report and the investigation. I've already planned a slight rework of class/super parsing, so I'm glad that someone has interest about this feature.

iborusik, 2020-01-17:
> Hi @iborusik!
> 
> Thanks for the report and the investigation. I've already planned a slight rework of class/super parsing, so I'm glad that someone has interest about this feature.
Hi @rerobika 

Thanks you for the support and quick response!

rerobika, 2020-03-05:
Hi @iborusik!

It took quite a long time, but https://github.com/jerryscript-project/jerryscript/pull/3598 is finally done. I've tested the enclosed test-case and the rework resolves this problem. :)

iborusik, 2020-03-05:
Hi @rerobika .  Many thanks for you fix :) I'll improve things a lot!))

rerobika, 2020-03-16:
Resolved via #3598.

closed 2020-03-16 by rerobika