jerryscript-project/jerryscript #3548: Why does JerryScript generate the all bytecode at once?

clover2123, 2020-02-07:
Hi there,

I have a question about compilation methodology of JerryScript.
It seems that JerryScript generates all bytecode for the whole source code at once (correct me if I'm wrong).
I think that it is inefficient for memory consumption because this approach holds the entire bytecode in memory but only part of source code is simultaneously executed during the runtime.
Is there any special reason for eager compilation?

And I have one more question.
Do you have any plan to implement the latest ECMA standard such as es10?
I ask you this because I also want to contribute to this project, so I'm searching for items now.

zherczeg, 2020-02-07:
Hi @clover2123,

JerryScript uses compact byte code (CBC) format, which is a small representation of the source code. Usually the CBC form consumes much less memory than the source code (even with source code minification), which we discard after the compilation. We know this is not strictly ES compatible, since you cannot use toString() on functions, but on low-end systems reducing memory consumption is more important than keeping the source code.

Do you have specific features in your mind which you are interested? Currently the only major ES2015 feature which was missing is Proxies (but it is in progress), and some medium features such as function names. We have already did some simple ES2016+ features, such as allowing comma at the end of function call, and async function declaration parsing (but no implementation for await), but the rest is open to take.

clover2123, 2020-02-10:
@zherczeg Really thank you for the kind reply. I do not have any specific plan now, so I'll take a look at the code one by one first.
Close this issue.

closed 2020-02-10 by clover2123