jerryscript-project/jerryscript #3554: Assertion (type == SCANNER_STREAM_TYPE_VAR) in scanner_is_context_needed

rwhitworth, 2020-02-13:
(my second bug report here. I apologize in advance for any mistakes)

###### JerryScript revision
5fdeb7c1d68e8c5209e186e1760bc703a723d588

###### Build platform
Debian GNU/Linux 10 (buster) (Linux 4.19.0-6-686-pae i686)

###### Build steps
```
CXX=afl-clang-fast++ CC=afl-clang-fast python tools/build.py --debug --clean --lto OFF
```
afl-clang-fast (and ++ version) built against `clang version 6.0.1-10 (tags/RELEASE_601/final)`

###### Test case
```
(function(){try{}catch(e){function e
```

###### Output
`Aborted`

###### Backtrace
Valgrind output:
```
==5914== Process terminating with default action of signal 6 (SIGABRT)
==5914==    at 0x4079382: raise (raise.c:51)
==5914==    by 0x40632B5: abort (abort.c:79)
==5914==    by 0x80EFD4D: jerry_port_fatal (default-fatal.c:30)
==5914==    by 0x8090F6D: jerry_fatal (jrt-fatals.c:63)
==5914==    by 0x8090FAB: jerry_assert_fail (jrt-fatals.c:87)
==5914==    by 0x809FBFF: scanner_is_context_needed (js-scanner-util.c:1605)
==5914==    by 0x80E0FD0: parser_parse_try_statement_end (js-parser-statm.c:1897)
==5914==    by 0x80E0FD0: parser_parse_statements (js-parser-statm.c:3123)
==5914==    by 0x8098472: parser_parse_function (js-parser.c:2415)
==5914==    by 0x80E04B4: parser_parse_function_statement (js-parser-statm.c:719)
==5914==    by 0x80E04B4: parser_parse_statements (js-parser-statm.c:2780)
==5914==    by 0x8098472: parser_parse_function (js-parser.c:2415)
==5914==    by 0x80E04B4: parser_parse_function_statement (js-parser-statm.c:719)
==5914==    by 0x80E04B4: parser_parse_statements (js-parser-statm.c:2780)
==5914==    by 0x8098472: parser_parse_function (js-parser.c:2415)
```

zherczeg, 2020-03-17:
Could you check that is this report still valid with the latest master? I get:

```
(function(){try{}catch(e){function e
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
Script Error: SyntaxError: Local variable is redeclared.
```

zherczeg, 2020-03-17:
Ah, it is ES5.1. Ok I can reproduce it.

PR #3618 (2020-03-17): Catch paramter must not be scanned as function in ES5.1
Fixes #3554
commit 1cfda262 (2020-03-17): Catch paramter must not be scanned as function in ES5.1 (#3618)

Fixes #3554

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com
closed 2020-03-17 by dbatyai