jerryscript-project/jerryscript #3555: Benchmark test for ES2015

clover2123, 2020-02-13:
Hello, I have a request about es6 benchmark suite.

The current process of implementing new es6 features looks not optimal because developer should manually adds test files. 

How about using es6-version `test262` (https://github.com/tc39/test262)?
test262 tests each language feature in a single file, so it would help finding bugs and eventually speed up development.
(I found that JerryScript already has test262 suite but it is for es5.1)

Thanks.

rtakacs, 2020-02-13:
@clover2123 Yes, ES6 tests of test262 can improve the testing process and it could add information how much ES6 features follow the standard. We planned to implement a ES6-test262 test driver in the near future, however if you are interested you can do it. Would you like to work on it?

clover2123, 2020-02-14:
@rtakacs actually, I have worked test262 of es6 version in my personal repo (https://github.com/clover2123/test262). But this version contains some wrong tests, so I'll futher update it and create a pr based on it if you are ok. Thanks.

ossy-szeged, 2020-02-14:
The idea is good to have a working test262 runner for ES2015 too. Unfortunately there isn't test262 release, branch or tag for ES2015. Of course we can pick a revision which we think it is close enough to ES2015 (as you picked one in your fork), but it can contain bugs, it can miss tests added later, etc.

I investigated master test262 and experimented how to run tests with jerry ~3 months before, but it changed a lot since ES5/ES6 and need more development to make it work properly. I'm going to continue this work in the near future. But I think it shouldn't conflict with the proposed ES2015 runner at all, they are completely different, test262 master is near 5 years older now. :)

Now tools/run-tests.py --test262 builds ES5 jerry, calls tools/runners/run-test-suite-test262.py
which clones ES5 test262, and run tests with the original runner in the test262 repository. It would be great to have a similar solution for ES2015, for example a --test262-es2015 option in run-tests.py script which calls a script to clone a given test262 revision and then run tests. Can we do it without modifying test262, just adding an exclude file and call its runner directly?

clover2123, 2020-02-17:
@ossy-szeged 
Yes, we can run ES6 test262 tests by cloning particular test262 version with some additional script files. 
But there are some issues that I want to discuss with you.
- test262 revision
By searching the log, [`fd44cd commit`](https://github.com/tc39/test262/commit/fd44cd73dfbce0b515a2474b7cd505d6176a9eb5) seems the last commit of ES6. How about picking this resivion for ES6 test? 
- test262 runner
it seems that original runner (test262.py) does not maintained properly now, so we should add ES6 runner manually. Other script files might be necessary too. I already have these additional scripts in my repo(https://github.com/clover2123/test262). Would it be fine to run test262 by cloning the particular test262 revesion first and copying the additional scripts into the cloned test262 directory before executing the runner?

clover2123, 2020-05-19:
#3667 is merged.
Close this issue.

closed 2020-05-19 by clover2123