jerryscript-project/jerryscript #3568: Implement missing Math functions from ES6 standard

rwalczyna, 2020-02-19:
To conform compatibility with ES6 some of the Math functions need to be implemented.

List of missing features:

- [ ] Math.acosh [ES v6, 20.2.2.3](https://www.ecma-international.org/ecma-262/6.0/#sec-math.acosh)
- [ ] Math.asinh [ES v6, 20.2.2.5](https://www.ecma-international.org/ecma-262/6.0/#sec-math.asinh)
- [ ] Math.atanh [ES v6, 20.2.2.7](https://www.ecma-international.org/ecma-262/6.0/#sec-math.atanh)
- [ ] Math.cbrt [ES v6, 20.2.2.9](https://www.ecma-international.org/ecma-262/6.0/#sec-math.cbrt)
- [ ] Math.clz32 [ES v6, 20.2.2.11](https://www.ecma-international.org/ecma-262/6.0/#sec-math.clz32)
- [ ] Math.cosh [ES v6, 20.2.2.13](https://www.ecma-international.org/ecma-262/6.0/#sec-math.cosh)
- [ ] Math.expm1 [ES v6, 20.2.2.15](https://www.ecma-international.org/ecma-262/6.0/#sec-math.expm1)
- [ ] Math.fround [ES v6, 20.2.2.17](https://www.ecma-international.org/ecma-262/6.0/#sec-math.fround)
- [ ] Math.hypot [ES v6, 20.2.2.18](https://www.ecma-international.org/ecma-262/6.0/#sec-math.hypot)
- [ ] Math.imul [ES v6, 20.2.2.19](https://www.ecma-international.org/ecma-262/6.0/#sec-math.imul)
- [ ] Math.log1p [ES v6, 20.2.2.21](https://www.ecma-international.org/ecma-262/6.0/#sec-math.log1p)
- [ ] Math.log10 [ES v6, 20.2.2.22](https://www.ecma-international.org/ecma-262/6.0/#sec-math.log10)
- [ ] Math.log2 [ES v6, 20.2.2.23](https://www.ecma-international.org/ecma-262/6.0/#sec-math.log2)
- [ ] Math.sinh [ES v6, 20.2.2.31](https://www.ecma-international.org/ecma-262/6.0/#sec-math.sinh)
- [ ] Math.tanh [ES v6, 20.2.2.34](https://www.ecma-international.org/ecma-262/6.0/#sec-math.tanh)

rwalczyna, 2020-02-19:
I made some research and it looks like JerryScript uses [_fdlibm_](http://www.netlib.org/fdlibm/) library (as [recommended](https://www.ecma-international.org/ecma-262/6.0/#sec-function-properties-of-the-math-object) by standard).
I can start with implementing missing logarithm functions basing on _fdlibm_.

ossy-szeged, 2020-02-19:
I already started this work in december, but unfortunately I didn't have time to finish it. In my working copy I added these functions to the codebase and they work fine with --jerry-libm=OFF (with standard libm). I'm going to upload a PR to merge it in a few days.

Once the skeleton is merged, it will be quite easy to add the missing fdlibm functions to jerry-libm one by one. It would be great if you could work on fdlibm implementations.

rwalczyna, 2020-02-19:
Okay, I will wait for your PR and after merge I can implement function from _fdlibm_.

rwalczyna, 2020-03-12:
As PR is already merged, I will start to work on porting fdlibm. I think that I will divide work into 4 parts:
1) logarithmic functions (log1p, log2, log10, expm1)
2) hyperbolic functions (cosh, sinh, tanh)
3) inverse hyperbolic functions(acosh, asinh, atanh)
4) other (cbrt)

rerobika, 2020-04-21:
@rwalczyna As I see all of these new Math functions are landed. Can we treat this issue resolved?

rwalczyna, 2020-04-21:
@rerobika Yes, I think that this issue can be finally closed :)

closed 2020-04-21 by rerobika