jerryscript-project/jerryscript #3578: Symbol properties have invalid attributes

rwalczyna, 2020-02-21:
As ES6 standard says in [19.4.2.2 - 19.4.2.14](https://www.ecma-international.org/ecma-262/6.0/#sec-symbol.hasinstance) Symbol properties should have the attributes set to **false**.
{ [[Writable]]: **false**, [[Enumerable]]: **false**, [[Configurable]]: **false** }.

###### Test code:
```js
["hasInstance", "isConcatSpreadable", "iterator", "keyFor",
"match", "prototype", "replace", "search", "species", "split",
"toPrimitive", "toStringTag", "unscopables"].forEach(function (v) {
  var desc = Object.getOwnPropertyDescriptor(Symbol, v)
  print("Symbol." + v + ":", desc.writable, desc.enumerable, desc.configurable);
});
```
###### Output:
```
Symbol.hasInstance: true false true
Symbol.isConcatSpreadable: true false true
Symbol.iterator: true false true
Symbol.keyFor: true false true
Symbol.match: true false true
Symbol.prototype: false false false
Symbol.replace: true false true
Symbol.search: true false true
Symbol.species: true false true
Symbol.split: true false true
Symbol.toPrimitive: true false true
Symbol.toStringTag: true false true
Symbol.unscopables: true false true
```
###### Expected output:
```
Symbol.hasInstance: false false false
Symbol.isConcatSpreadable: false false false
Symbol.iterator: false false false
Symbol.keyFor: true false true
Symbol.match: false false false
Symbol.prototype: false false false
Symbol.replace: false false false
Symbol.search: false false false
Symbol.species: false false false
Symbol.split: false false false
Symbol.toPrimitive: false false false
Symbol.toStringTag: false false false
Symbol.unscopables: false false false
```

As I checked, this bug comes from #3490 . I think that **INTRINSIC_PROPERTY** macro should have flag parameter set to **ECMA_PROPERTY_FIXED** .

But this will also change behaviour of other properties created with that macro: 
* _Array.prototype.values_
* _Array.prototype[Symbol.iterator]_

These properties should have { [[Writable]]: **true**, [[Enumerable]]: **false**, [[Configurable]]: **true** }, as stated in [ES 6 17](https://www.ecma-international.org/ecma-262/6.0/#sec-ecmascript-standard-built-in-objects)
So it looks like other macro is needed, i.e. **INTRINSIC_PROPERTY_CONFIGURABLE.**

Does it sound reasonable? If yes, then I could implement this and add some tests.

dbatyai, 2020-02-21:
We could also add a 3rd argument to the `INTRINSIC_PROPERTY` macro, which can be used to specify property attributes. Most other macros already use this approach.

rerobika, 2020-02-21:
+1 for the suggested 3rd argument.

PR #3581 (2020-02-24): Fix Symbol properties descriptors.
Symbol properites descriptors had invalid values

Fixes #3578

JerryScript-DCO-1.0-Signed-off-by: Rafal Walczyna r.walczyna@samsung.com
commit 20300189 (2020-02-24): Fix Symbol properties descriptors. (#3581)

Symbol properites descriptors had invalid values

Fixes #3578

JerryScript-DCO-1.0-Signed-off-by: Rafal Walczyna r.walczyna@samsung.com
closed 2020-02-24 by dbatyai