jerryscript-project/jerryscript #3589: Memory Access Violation in ecma_compare_ecma_non_direct_strings

niebardzo, 2020-02-25:
###### JerryScript revision
9b393ee2ea07ef4e7971e72637e46923dfc15e6f

###### Build platform
Ubuntu 18.04.4 LTS (Linux 4.15.0-76-generic x86_64)

###### Build steps
```
./tools/build.py --clean --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Output

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==16395==ERROR: AddressSanitizer: SEGV on unknown address 0x5f5e0ff4 (pc 0x0813fa83 bp 0xffc3b0c8 sp 0xffc3b0b0 T0)
==16395==The signal is caused by a READ memory access.

    #0 0x813fa82 in ecma_compare_ecma_non_direct_strings /root/work/libfuzzer/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:1819:20
    #1 0x81476d3 in ecma_find_named_property /root/work/libfuzzer/jerryscript/jerry-core/ecma/base/ecma-helpers.c:707:13
    #2 0x8181593 in ecma_op_object_get_own_property /root/work/libfuzzer/jerryscript/jerry-core/ecma/operations/ecma-objects.c:236:33
    #3 0x818209c in ecma_op_object_get_property /root/work/libfuzzer/jerryscript/jerry-core/ecma/operations/ecma-objects.c:354:32
    #4 0x818209c in ecma_op_object_has_property /root/work/libfuzzer/jerryscript/jerry-core/ecma/operations/ecma-objects.c:409:30
    #5 0x8235efa in ecma_op_put_value_lex_env_base /root/work/libfuzzer/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:275:13
    #6 0x81e7e8d in vm_loop /root/work/libfuzzer/jerryscript/jerry-core/vm/vm.c:3874:43
    #7 0x81de804 in vm_execute /root/work/libfuzzer/jerryscript/jerry-core/vm/vm.c:4194:37
    #8 0x81de12f in vm_run /root/work/libfuzzer/jerryscript/jerry-core/vm/vm.c:4302:10
    #9 0x81de3b7 in vm_run_global /root/work/libfuzzer/jerryscript/jerry-core/vm/vm.c:320:10
    #10 0x81235c7 in jerry_run /root/work/libfuzzer/jerryscript/jerry-core/api/jerry.c:593:24
    #11 0x811b483 in main /root/work/libfuzzer/jerryscript/jerry-main/main-unix.c:748:23
    #12 0xf7cd4e80 in __libc_start_main (/lib32/libc.so.6+0x18e80)
    #13 0x8071eb1 in _start (/root/work/libfuzzer/jerryscript/build/bin/jerry+0x8071eb1)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/work/libfuzzer/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:1819:20 in ecma_compare_ecma_non_direct_strings
==16395==ABORTING
```

rerobika, 2020-02-25:
Hi @niebardzo!

Please enclose the related test case that caused this error.

niebardzo, 2020-02-26:
Hello @rerobika ,

Sorry, I forgot to attach it. Here is the test case:
```
[this,000000000,this,99999999=9999999]
```

PR #3594 (2020-03-16): Fix assignment validation in parser_append_binary_token
Fixes #3589

JerryScript-DCO-1.0-Signed-off-by: Adam Szilagyi aszilagy@inf.u-szeged.hu
commit 7ea93aff (2020-03-16): Fix assignment validation in parser_append_binary_token (#3594)

Fixes #3589

JerryScript-DCO-1.0-Signed-off-by: Adam Szilagyi aszilagy@inf.u-szeged.hu
closed 2020-03-16 by rerobika